adafruit/Adafruit_SPIFlash #10: example fatfs_format fails for Adafruit Metro M4

MrBryonMiller, 2019-03-06:
Using an Adafruit Metro M4 and while trying to troubleshoot this Adafruit forum issue https://forums.adafruit.com/viewtopic.php?f=60&t=148446#p733537 I ran into problems when using commit 1.1.0 (commit 1.0.8 worked fine).  

To simplify I tried just fatfs_format and found it to fail with the following info :

> Adafruit SPI Flash FatFs Format Example
> Flash chip JEDEC ID: 0x0
> !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
> This sketch will ERASE ALL DATA on the flash chip and format it with a new filesystem!
> Type OK (all caps) and press enter to continue.
> !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
> Partitioning flash with 1 primary partition...
> Error, f_fdisk failed with error code: 1
>

ladyada, 2019-03-06:
metro m4 is a QSPI - please use that!

ladyada, 2019-03-06:
(or you can define the SPI port properly  - the default almost certainly wont work, you need to bitbang it)

MrBryonMiller, 2019-03-06:
You're right. I had oversimplified my problem.  My actual difficulties show up in a much more complicated program.  I'll try to simplify and post an example that shows the problem.

MrBryonMiller, 2019-03-06:
OK - here's what I should have said: The fatfs_format example **from the QSPI repo** fails to format with the same error message I showed above (except that example does not show the Flash chip JEDEC ID).  I could open an issue there except nothing there has changed.  The only change I can find is in this repo. Please let me know if it would be more appropriate to open issue in QSPI repo.

ladyada, 2019-03-06:
ok how about you try getting this library version working first? you would bitbang the SPI interface

ladyada, 2019-03-06:
(i wont have time to get to this for a bit)

MrBryonMiller, 2019-03-07:
@ladyada wrote 

> ok how about you try getting this library version working first? you would bitbang the SPI interface

So I changed the SPI interface to software rather than hardware and, sure enough, the example formatted the flash as a file system.  So I tried my more complex program and it was also able to to format the flash.  But when I told it to make a directory it failed to - as a matter of fact the program hung up.  My **guess** is because making a directory uses writeBuffer and that is one of the routines that had an interface change.  

I think the next step would be either to go through the routines in QSPI repo and make sure the definitions are the same as used in latest version of this repo OR back the interface changes back out of this repo.  Since I don't understand why the interface for buffer was changed to 'const' I'm not at all sure which way is the right way.

> (i wont have time to get to this for a bit)

As long as you get to it eventually I'm not in a particular hurry since I will just continue to use the prior version.

ladyada, 2019-03-08:
ok i tried the fatfs_circuitpython example, works great, i commited some text changes

ladyada, 2019-03-08:
ok i commited a version of the example here that bitbangs the QSPI chip as SPI - can y'all try it out?

MrBryonMiller, 2019-03-08:
Yes - I tried fatfs_circuitpython and it worked.  Took me awhile because I first had to look up how to install CP since it won't work without it being initially installed.

Then I made similar edits to two other examples (fatfs_format and fatfs_fullusage) And they both worked.  I then tried my "Arduino DOS" program and it also worked! (I swear it did not yesterday but musta been a brain fart).

So all is good with me IF I use Software SPI.   But I think in order to get QSPI speed will still need for this repo and QSPI_Flash to get into sync? So original Issue still stands? Is this the way you see it?

ladyada, 2019-03-08:
original issue stands but we have a nice workaround at least and its probably not *much* slower since we aren't DMA'ing much. if you can submit a PR that adds the pinouts for the other examples - that would be really helpful for me. at least then its somewhat resolved & well move the issue to QSPI repo

MrBryonMiller, 2019-03-09:
See PR #11 - hopefully it is what you wanted.  Are you going to move this issue to QSPI repo or do you want me to?

ladyada, 2019-03-09:
thanks for the PR - yeah can you re-open this in QSPI? ill close it here!

closed 2019-03-09 by ladyada

MrBryonMiller, 2019-03-09:
So here's the problem with working with a retired engineer (read an old guy) - I forgot that two days ago I found that @Macrosii had open a very similar issue in the QSPI repo where it is issue 5.  I even commented on it two days ago.  I think that issue removes the need for opening anything new there.

ladyada, 2019-03-09:
lolll! ok thanks :) this is great