adafruit/Adafruit_SPIFlash #19: Using Adafruit_SPIFlash on the Metro M4

GeneSu730, 2019-08-02:
Dear All:

I am doing a project for school involving the Metro M4. One of the features of the M4 I am exploiting is its data logging ability with the QSPI flash memory. I actually declared nominal success on this project back in February with the exception of some bugs, mostly an inability to consistently extract the data file from the Flash drive.
I was writing a manual for the operation of my system when I learned that the QSPI library had just been depreciated and that the Adafruit_SPIFlash library had been heavily revised. So I downloaded the newest versions of the Adafruit_SPIFlash (3.1.1) and SDFat (1.1.0) libraries. After some tinkering, I ran the SDFat_Format.ino sketch to re-format my Metro M4's flash memory. At the end, I got this message:

`Error, failed to mount newly formatted filesystem!`

Are there any configuration settings I have to first change before I run the sketch? I know the sketch was primarily written for the M0. However, the M4 uses a QSPI protocol to communicate with its on-board memory. I see these settings at the beginning of the file.

```
#if defined(__SAMD51__) || defined(NRF52840_XXAA)
  Adafruit_FlashTransport_QSPI flashTransport(PIN_QSPI_SCK, PIN_QSPI_CS, PIN_QSPI_IO0, PIN_QSPI_IO1, PIN_QSPI_IO2, PIN_QSPI_IO3);
#else
  #if (SPI_INTERFACES_COUNT == 1)
    Adafruit_FlashTransport_SPI flashTransport(SS, &SPI);
  #else
    Adafruit_FlashTransport_SPI flashTransport(SS1, &SPI1);
  #endif
#endif

Adafruit_SPIFlash flash(&flashTransport);

// file system object from SdFat
FatFileSystem fatfs;
```
How should I ensure that `__SAMD51__ `is defined? Do I `#define` it manually?

In my original data logging code, the flash type is Adafruit_QSPI_GD25Q. The FAT system is Adafruit_W25Q16BV_Fatfs. How can I ensure that these options are set properly in the new libraries?

I got this error from working in the school lab last night. I wasn't allowed a lot of time to attend to this issue. I know that there have been a lot of postings on the Adafruit forum about the M4 and the SPIFlash library so I apologize if my question is already answered. I will try to go to the lab either today or Monday and will post any updates. 
I would appreciate it if someone could help me. A big bonus if the data file is no longer corrupted on any write.

Also, I was checking out this website address:

https://learn.adafruit.com/adafruit-fea ... -spi-flash

So this represents the learning website for using the SPIFLASH library. Here is the link to the library.

https://github.com/adafruit/Adafruit_SP ... master.zip

Now, the website gives this list of sketches that I will be able to run:
fatfs_circuitpython
fatfs_datalogging
fatfs_format
fatfs_full_usage
fatfs_print_file
flash_erase

None of the sketches that start with fatfs are in the zip file. Will the SDFat_format sketch really encompass QSPI flash memories and the Metro M4?

I got this error from working in the school lab last night. I wasn't allowed a lot of time to attend to this issue. I know that there have been a lot of postings on the Adafruit forum about the M4 and the SPIFlash library so I apologize if my question is already answered. I will try to go to the lab either today or Monday and will post any updates. 
I would appreciate it if someone could help me. A big bonus if the data file is no longer corrupted on any write.

S2Doc, 2019-08-02:
__SAMD51__ should be automatically defined by the Metro M4 board file.  Are you using the Arduino IDE?

You will have to have the following files in the same folder as your main .ino program file:

- diskio.h
- ff.h
- ff.c
- ffconf.h

You will want to have the code fragment you quoted in your post as part of your sketch.

**This is the routine I use to initialize and format the SPI flash:**
boolean FormatSPIflash()  {
  //Format SPI/QSPI FLASH memory as SD
  Serial.println(F("**********************************************"));
  Serial.println(F("**********************************************"));
  Serial.println(F("**********************************************"));
  Serial.println(F("FormatSPIflash:  Formatting QSPI FLASH as FAT volume..."));
  // Initialize flash library and check its chip ID.
  if (!flash.begin()) {
    Serial.println("FormatSPIflash:   ERROR, failed to initialize SPI flash chip!");
    return false;
  }
  Serial.print(F("Flash chip JEDEC ID: 0x")); Serial.println(flash.getJEDECID(), HEX);
  Serial.println(F("This may take up to ~30 seconds until the flash erase is finished."));
  Serial.println(F("A success or error message will be printed when complete."));
  // Call fatfs begin and passed flash object to initialize file system
  if ( fatfs.begin(&flash) ){
    if ( !fatfs.wipe(&Serial) ) {
      Serial.println(F("FormatSPIflash:  Failed to wipe SPI FLASH chip"));
    }
  }
  else {
    Serial.println(F("FormatSPIflash:  Failed to init filesystem"));
    Serial.println(F("FormatSPIflash:  Creating and formatting FAT filesystem (this may take many seconds)..."));
    // Make filesystem.
    uint8_t buf[512] = {0};          // Working buffer for f_fdisk function.    
    FRESULT r = f_mkfs("", FM_FAT | FM_SFD, 0, buf, sizeof(buf));
    if (r != FR_OK) {
      Serial.print(F("FormatSPIflash:  Error, f_mkfs failed with error code: ")); Serial.println(r, DEC);
      return false;
    }
    Serial.println(F("FormatSPIflash:  Formatted flash!"));
  }
  // Must reinitialize after wipe.
  if (!fatfs.begin(&flash)) {
    Serial.println(F("FormatSPIflash:  Error, failed to mount newly formatted filesystem!"));
    return false;
  }
  // Done!
  Serial.println(F("FormatSPIflash:  Flash chip successfully formatted with new empty filesystem!"));
 gSPIflashIsReady = true;
  return true;
}

**also**

void InitializeSPIflash()  {
 // Initialize flash library and check its chip ID.
  gSPIflashIsReady = true;
  if (!flash.begin()) {
    Serial.println("InitializeSPIflash:  Error, failed to initialize flash chip!");
    gSPIflashIsReady = false;
  }
  if (gSPIflashIsReady ) {
    if (!fatfs.begin(&flash)) {
      Serial.println(F("InitializeSPIflash:  Error, fatfs,begin() failed to initialize file system"));
    }
  }
  Serial.print("InitializeSPIflash:  Mounted filesystem!  Flash chip JEDEC ID: 0x"); Serial.println(flash.getJEDECID(), HEX);
  gSPIflashIsReady = true;
}

**This is how I create and write files:**
boolean SPIflashWriteString(String fileName, String dataStr) {
  if (!gSPIflashIsReady) InitializeSPIflash();
  if (!gSPIflashIsReady)  {
    Serial.println(F("SPIflashWriteString:  InitializeSPIflash failed!   <<<<<<<"));
    return false;
  }
  //Create the file for writing
  char fName[200];
  fileName.toCharArray(fName, sizeof(fName));
  File dataFile = fatfs.open(fName, FILE_WRITE);
  if (!dataFile) {
    Serial.println(F("SPIflashWriteString:  Error, failed to create/open file for writing!"));
    return false;
  }
  Serial.println(F("Opened file for writing/appending..."));
// if the file is available, write to it:
  if (dataFile) {
    dataFile.seek(0);   //Go to beginning of file
    dataFile.print(dataStr);
    dataFile.close();
    return true;
  }
  else  {
    Serial.println(F("SPIflashWriteString:  Error, failed to create/open file for writing!"));
    return false;
  }
}

**I read files using this:**
String SPIflashReadString(String fileName)  {
  File dataFile;
  char fName[200];
  if (!gSPIflashIsReady) InitializeSSPIflash();
  if (!gSPIflashIsReady)  {
    Serial.println(F("SPIflashReadString:  InitializeSD() failed!   <<<<<<<"));
    return kEmptyString;
  }
  fileName.toCharArray(fName, sizeof(fName));
  if (!fatfs.exists(fName)) {
    Serial.println(kEmptyString+F("SPIflashReadString: ")+fileName+F(" does not exist!  **"));
    return kEmptyString;
  }
  dataFile = fatfs.open(fName);
  if (!dataFile) {
    Serial.println(kEmptyString+F("SPIflashReadString failed to open file: ")+fileName);
    return kEmptyString;
  }
  int kMaxBytes = 1024;     //Max file length (arbitrary)
  char buf[kMaxBytes]; 
  char ch;
  int i = 0;
  //Read data until there is none left
  while (dataFile.available() && (i<kMaxBytes)) {
    ch = dataFile.read();
    buf[i] = ch;
    i++;
  }
  buf[i] = 0;   //Append terminating null
  dataFile.close();
  return String(buf);
}

I hope this helps.

GeneSu730, 2019-08-02:
1. I am using the Arduino IDE. I installed it in the school lab computer in December. Because of issues with the LAN network, I have not been able to connect the computer online since March. Would that pose a problem? I have been trying to find a new LAN port in our newly renovated room and hope to resolve this next week.

2. I do have these 4 files in the SdFat_Format folder:
diskio.h
ff.h
ff.c
ffconf.h

3. It seems that **FormatSPIFlash** has the exact same algorithm as the **SDFat_Format.ino** sketch. Remember. I ran that sketch and got the error message. I will copy and paste it here:

```
void setup() {
  // Initialize serial port and wait for it to open before continuing.
  Serial.begin(115200);
  while (!Serial) {
    delay(100);
  }
  Serial.println("Adafruit SPI Flash FatFs Format Example");

  // Initialize flash library and check its chip ID.
  if (!flash.begin()) {
    Serial.println("Error, failed to initialize flash chip!");
    while(1);
  }
  Serial.print("Flash chip JEDEC ID: 0x"); Serial.println(flash.getJEDECID(), HEX);

  // Wait for user to send OK to continue.
  Serial.setTimeout(30000);  // Increase timeout to print message less frequently.
  do {
    Serial.println("!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!");
    Serial.println("This sketch will ERASE ALL DATA on the flash chip and format it with a new filesystem!");
    Serial.println("Type OK (all caps) and press enter to continue.");
    Serial.println("!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!");
  }
  while (!Serial.find("OK"));

  // Call fatfs begin and passed flash object to initialize file system
  if ( fatfs.begin(&flash) ){
    if ( !fatfs.wipe(&Serial) ) {
      Serial.println("Failed to wipe");
    }
  }else {
    Serial.println("Failed to init filesystem");
    Serial.println("Creating and formatting FAT filesystem (this takes ~60 seconds)...");

    // Make filesystem.
    uint8_t buf[512] = {0};          // Working buffer for f_fdisk function.    
    FRESULT r = f_mkfs("", FM_FAT | FM_SFD, 0, buf, sizeof(buf));
    if (r != FR_OK) {
      Serial.print("Error, f_mkfs failed with error code: "); Serial.println(r, DEC);
      while(1);
    }
    Serial.println("Formatted flash!");
  }

  // Must reinitialize after wipe.
  if (!fatfs.begin(&flash)) {
    Serial.println("Error, failed to mount newly formatted filesystem!");
    while(1) delay(1);
  }

  // Done!
  Serial.println("Flash chip successfully formatted with new empty filesystem!");
}
```

4. It seems that the **InitializeSPIflash** subroutine is replicated in the setup of the **SdFat_datalogging.ino** sketch.

5. I guess my question is whether I have to do anything to the configuration variables at the start of the **SdFat_format** sketch.

6. Please also note I am doing this because the **Adafruit_QSPI** library has been depreciated. I want to make sure that the new **Adafruit_SPIFlash** library works with the Metro M4.

ladyada, 2019-08-02:
yes it definitely works, you need to have the latest adafruit sdfat library, and you can load circuitpython to format the disk

GeneSu730, 2019-08-02:
Can you give me a link to the latest Adafruit SDFat library? I downloaded
an SDFat library but I am not so sure if it is the correct one.
Also, is Circuit Python necessary?

ladyada, 2019-08-02:
https://github.com/adafruit/SdFat
its not necessary, but it can help because its an easy way to manipulate the filesys

GeneSu730, 2019-08-02:
I am going to check this out on Monday in the school lab and then get back
to you.

GeneSu730, 2019-08-02:
I noticed that the new SdFat library that you sent me has this extra file:
SdSpiSAMD.cpp
Was that written by Adafruit?

ladyada, 2019-08-02:
yes

GeneSu730, 2019-08-05:
I hooked up the school computer to a LAN port. It is now connected. I updated Arduino to 1.8.9. I also updated the boards drivers. I have the latest version of Adafruit_SPIFlash and Adafruit_SdFat. When I ran the formatter, this is the message I got:

```
Adafruit SPI Flash FatFs Format Example
Flash chip JEDEC ID: 0xC84015
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
This sketch will ERASE ALL DATA on the flash chip and format it with a new filesystem!
Type OK (all caps) and press enter to continue.
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
.
Flash chip successfully formatted with new empty filesystem!
```

So far, so good. Now I modified my program to have the correct headers. Here are two cut and pastes:

```

/************************ FILE-HANDLING LIBRARIES ************************/

#include <SPI.h>
#include <SdFat.h>
#include <Adafruit_SPIFlash.h>

/************************ CONFIGURATION CONSTANTS FOR FILE-HANDLING ************************/

// Configuration of the flash chip pins and flash fatfs object.
// You don't normally need to change these if using a Feather/Metro
// M0 express board.
#if defined(__SAMD51__) || defined(NRF52840_XXAA)
  Adafruit_FlashTransport_QSPI flashTransport(PIN_QSPI_SCK, PIN_QSPI_CS, PIN_QSPI_IO0, PIN_QSPI_IO1, PIN_QSPI_IO2, PIN_QSPI_IO3);
#else
  #if (SPI_INTERFACES_COUNT == 1)
    Adafruit_FlashTransport_SPI flashTransport(SS, &SPI);
  #else
    Adafruit_FlashTransport_SPI flashTransport(SS1, &SPI1);
  #endif
#endif

/************************ VARIABLES FOR FILE-HANDLING ************************/

Adafruit_SPIFlash flash(&flashTransport);

// file system object from SdFat
FatFileSystem fatfs;

// Configuration for the datalogging file:
File dataFile;
#define FILE_NAME      "data.csv"
```

```
/************************* DATA RECORDING ****************************/

      // Open the datalogging file for writing.  The FILE_WRITE mode will open
      // the file for appending, i.e. it will add new data to the end of the file.
      dataFile = fatfs.open(FILE_NAME, FILE_WRITE);
      // Check that the file opened successfully and write a line to it.
  
      if (dataFile) {
        // Write the data reading to the file.  You can use all the same print functions
        // as if you're writing to the serial monitor.          
        dataFile.println("----------------");
        for (CycleCount = 0; CycleCount < MaxCycle; CycleCount++)
        {
          //dataFile.print(FbkErr[CycleCount], 4);
          dataFile.print(FbkErr[CycleCount], 5);
          dataFile.print(",");          
          dataFile.println(EstFrs[CycleCount], 4);          
        }
      
        // Finally close the file when done writing.  This is smart to do to make
        // sure all the data is written to the file.
        dataFile.close();
        Serial.println("Data stored in file data.csv");
      }
      else {
        Serial.println("Failed to open data file for writing!");
      }
```

I double click the reset button and drag **adafruit-circuitpython-metro_m4_express-en_US-4.1.0.uf2** to the drive and see this mess:

[image]

Is there anything else I need to do?

ladyada, 2019-08-05:
yyeah ive seen this but not sure why, maybe related to the 1/2 FAT issue. please format it with the OS formatter? does that help

GeneSu730, 2019-08-05:
Which sketch is the OS formatter?

ladyada, 2019-08-05:
no, we mean your operating system.

GeneSu730, 2019-08-05:
You want me to format my entire Windows 10 operating system?

ladyada, 2019-08-05:
no, format the drive using the operating system

GeneSu730, 2019-08-05:
Let me see if that works... I'll get back to you by tomorrow at the latest.

GeneSu730, 2019-08-05:
Formatting the drive in Windows does the trick. Thank you!

closed 2019-08-09 by GeneSu730