adafruit/Adafruit_SPIFlash #25: Did not work with github.com/stm32duino/Arduino_Core_STM32 and SPI2

rtek1000, 2019-09-26:
If anyone tries to use the SPI2 port (Blue Pill, STM32F103C8T6), only core github.com/rogerclarkmelbourne/Arduino_STM32 worked.

Did not worked (Arduino_Core_STM32):
```
#define MOSI2 PB15
#define MISO2 PB14
#define SCLK2 PB13
#define SS2 PB12

//SPIClass(uint8_t mosi, uint8_t miso, uint8_t sclk, uint8_t ssel)
SPIClass SPI_2 = SPIClass(MOSI2, MISO2, SCLK2, SS2);

Adafruit_FlashTransport_SPI flashTransport(SS2, &SPI_2);

void setup() {
  SPI_2.begin();
 //  remaining code below
}
```

It worked (Arduino_STM32):
```
#define SS2 PB12
SPIClass SPI_2(2);
Adafruit_FlashTransport_SPI flashTransport(SS2, &SPI_2);
```

Reported here for future research:
https://github.com/stm32duino/Arduino_Core_STM32/issues/669

hathach, 2019-09-26:
@rtek1000 I dont understand why would you open the issue then close it right away ?

rtek1000, 2019-10-02:
@hathach Because while it is just a report, it can be a waste of time. But if it's really important, other users will report the same issue and open an issue in more detail.

rtek1000, 2019-10-02:
I found something interesting, with the "Arduino_STM32" core the flash_info sketch reads 0xFF after chip erase. But with "Arduino_Core_STM32" core, the first byte is 0x7F for all pages, this may explain why it doesn't work on this core.

rtek1000, 2019-10-02:
Seems to be just an SPI port speed issue, for some reason after slowing down the 0x7F is no longer found:

Added to main code after flash.begin();

```C++
  flashTransport.setClockSpeed(200*1000000UL); // 200MHz: fail (1st: 0x7F)
```

```C++
  flashTransport.setClockSpeed(1*1000000UL); // 1MHz: ok (all: 0xFF)
```

That's it, the FAT part is working now, just modify the speed.

Added to main code after flash.begin();
```C++
flashTransport.setClockSpeed(1*1000000UL); // 1MHz: ok (all: 0xFF)
```

Scketch SdFat_format results on serial monitor:

> Adafruit SPI Flash FatFs Format Example
> Flash chip JEDEC ID: 0xEF4017
> !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
> This sketch will ERASE ALL DATA on the flash chip and format it with a new filesystem!
> Type OK (all caps) and press enter to continue.
> !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
> Creating and formatting FAT filesystem (this takes ~60 seconds)...
> Formatted flash!
> Flash chip successfully formatted with new empty filesystem!
> 

Scketch SdFat_full_usage results on serial monitor:

> Adafruit SPI Flash FatFs Full Usage Example
> Flash chip JEDEC ID: 0xEF4017
> Mounted filesystem!
> Test directory not found, creating...
> Created directory!
> Creating deep folder structure...
> Creating /test/foo/bar
> Created directory!
> Creating /test/foo/baz
> Created directory!
> Opened file /test/test.txt for writing/appending...
> Wrote to file /test/test.txt!
> First line of test.txt: Hello world!
> 
> Total size of test.txt (bytes): 57
> Current position in test.txt: 14
> Available data to read in test.txt: 43
> File name: use getName()
> Is file a directory? No
> Entire contents of test.txt:
> Hello world!
> Hello number: 123
> Hello hex number: 0x7B
> Listing children of directory /test:
> - foo (directory)
> - test.txt
> Deleting /test/foo/test2.txt...
> Deleted file!
> Deleting /test directory and everything inside it...
> Test directory was deleted!
> Finished!
>

rtek1000, 2019-10-02:
Note: I reopened the issue because the auto configured SPI port speed may be too high

rtek1000, 2019-10-02:
In the flash_devices.h file, replacing 133 (MHz) with 35 (MHz) the format and full_usage examples worked without having to add setClockSpeed after begin:

(Note: with 40MHZ no longer works)

```C++
// Settings for the Winbond W25Q64JV-IQ 8MiB SPI flash. Note that JV-IM has a different .memory_type (0x70)
// Datasheet: http://www.winbond.com/resource-files/w25q64jv%20revj%2003272018%20plus.pdf
#define W25Q64JV_IQ {\
    .total_size = (1 << 23), /* 8 MiB */ \
    .start_up_time_us = 5000, \
    .manufacturer_id = 0xef, \
    .memory_type = 0x40, \
    .capacity = 0x17, \
    .max_clock_speed_mhz = 35, \
    .quad_enable_bit_mask = 0x02, \
    .has_sector_protection = false, \
    .supports_fast_read = true, \
    .supports_qspi = true, \
    .supports_qspi_writes = true, \
    .write_status_register_split = false, \
    .single_status_byte = false, \
}
```

hathach, 2019-10-02:
> In the flash_devices.h file, replacing 133 (MHz) with 35 (MHz) the format and full_usage examples worked without having to add setClockSpeed after begin:
>
> (Note: with 40MHZ no longer works)
>
> ```c++ [...]

You shouldn't edit this, this is the max speed that the flash device can work with by its specification. Although stm32 couldn't run reliably at that speed, it doesn't mean other mcu should run at lower speed with this flash device. Anyway, I admit, we have no stm32 core support now, and troubleshoot this issue is currently beyond my  ability.

rtek1000, 2019-10-02:
Well, I don't know, I opened a datasheet that says only 104MHz for this 4017 memory type, so if anyone wants to try using this W25Q64, they may have to set the SPI port speed to work.

rtek1000, 2019-10-02:
I found an interesting quote, perhaps better explain the SPI port speed situation on this microcontroller:

> The STM32F103C8 Blue Pill board runs at 72 MHz. So APB2 is at 72 MHz and APB1 is at 36 MHz. This means that when initializing SPI1, the pins can be set up to their maximum of 50 MHz. But when initializing SPI2, the pins can only be set up at 10 MHz maximum (I am not sure about this but it seems to be the safer option). The other thing to consider that when setting the SPI baud rate using the prescaler, the reference clock speed is the bus speed. For example, for SPI1 on APB2, SPI_BaudRatePrescaler_64 gives 72/64 = 1.125 MHz. But if you try to do the same for SPI2 on APB1, SPI_BaudRatePrescaler_64 gives 36/64 = 0.56 MHz.

Source: https://learnbuildshare.wordpress.com/about/stm32/using-spi-as-master/

As the library does not officially support this core, so only the user has manual SPI port speed control, as I had reported, worked at 35MHz, and if the port can only operate up to 50MHz in theory, then 35MHz may be within expected.

rtek1000, 2019-10-02:
>  rtek1000 commented:
> Would it be a problem with the core?
> 
> Why does the other core work with SPI set to 133MHz and this core here only works at 35MHz or below?

>  fpistm commented:
> Roger's core use SPI with DMA.
> An issue is already opened to make some SPI optimizations

https://github.com/stm32duino/Arduino_Core_STM32/issues/669

closed 2019-10-02 by rtek1000