adafruit/Adafruit_SPIFlash #28: [F-RAM] Would it be possible to add FRAM memory support?

rtek1000, 2019-10-02:
I am trying to use a model FM25L04G F-RAM as a buffer for a flash memory to avoid unnecessary flash use, the flash being used only when there is definite data to be saved, as the F-RAM theoretically allows unlimited cycles reading and writing.

Based on the flash_devices.h file, I edited a new definition:

```C++
// Settings for the RAMTRON FM25L04G 4Kbit SPI F-RAM.
// Datasheet: https://www.dataman.com/media/datasheet/Ramtron/FM25L04.pdf
#define FM25L04  {\
    .total_size = (1 << 9), /* 512 bytes */ \
    .start_up_time_us = 10000, /* 10 ms */ \ 
    .manufacturer_id = 0x00, \
    .memory_type = 0x00, \
    .capacity = 0x09, \
    .max_clock_speed_mhz = 14, \
    .quad_enable_bit_mask = 0x00, \
    .has_sector_protection = false, \
    .supports_fast_read = false, \
    .supports_qspi = false, \
    .supports_qspi_writes = false, \
    .write_status_register_split = false, \
    .single_status_byte = true, \
}
```
This memory has few commands, and are of the same standard as Flash W25Q64, but has no JEDEC ID, how can I force the parameters described in the flash_devices.h file?

[image]

ladyada, 2019-10-03:
thats odd, you're sure there's no JEDEC? its supposed to be standard :/

hathach, 2019-10-03:
> I am trying to use a model FM25L04G F-RAM as a buffer for a flash memory to avoid unnecessary flash use, the flash being used only when there is definite data to be saved, as the F-RAM theoretically [...]

I checked the data sheet, this device indeed doesn't support JEDEC ID command. Which means there is no way for the library to dynamically detect it. You probably need to hard code it e.g skipping JEDEC, and make sure it is the only flash device in used.

sslupsky, 2019-10-10:
That must be an odd part.

If you use a Fujitsu M85RS1M device, it has an ID with op-code 1001 1111B.

The Fujitsu part is much larger, 1MBit (128K x 8).

hathach, 2020-07-24:
head up, just want you guys to know that we are planning to add FRAM support for those 1Mb, & 2Mb parts.

closed 2020-08-18 by hathach

hathach, 2020-08-20:
FRAM is supported, by detection is only possible with device responded to RDID. For your flash device, try to use 0xFFFFFF or 0x000000 and provide your own flash list to match.

PR #63 (2020-08-18): Support FRAM also vastly improve speed using DMA
close #28 
- Added support for Fujitsu FRAM 128/256/512 KB : MB85RS1MT/MB85RS2MTA/MB85RS4MT. The support is transparent to user sketch, just make sure the flash transport is declared correctly, the FRAM should be detected automatically.
- Added flash_speedtest example
- Added `void setIndicator(int pin, bool state_on)`
- Improve SPI speed by using DMA transfer for
  - DMA SPI read and write for nrf52
  - DMA SPI write for SAMD M0/M4. There is issue with DMA SPI read on these platform. There will be separated PR/issue later on
-  Allow different  [... cut]