adafruit/Adafruit_SPIFlash #29: Feature request:  Support SPI NAND flash devices

sslupsky, 2019-10-04:
Is there any interest in updating the library to support SPI NAND devices?  These devices are quite a bit larger than the NOR devices currently supported and are available in sized of several GBytes.

Here is a link to a Toshiba data sheet:
https://media.digikey.com/pdf/Data%20Sheets/Toshiba%20PDFs/TC58CVG2S0HxAIx_Rev1.1_2016-11-08.pdf

dhalbert, 2019-10-04:
NAND flash is pretty different than the NOR flash SPI devices that this library supports. NOR flash is random-access: you can read individual bytes or words. NAND flash is page or block oriented: you read a whole page at a time.

I think it would make sense to start a different library rather than add to this one.

NAND flash has things like ECC bits exposed, and often expects to be used with a controller chip. An SD card is NAND flash with a front end. So in a sense, we already have support for NAND flash, it's just at a higher level (not raw access).

sslupsky, 2019-10-04:
Ok, thanks for the feedback.  I had noticed that there was a "FlashCache" layer and wondered if it could be used to for writing the flash pages and was wondering if the library might be heading in the NAND direction or not.

closed 2019-10-04 by sslupsky