adafruit/Adafruit_SPIFlash #32: W25Q128JV_SQ and W25Q128JV_PM don't work

Casey10110, 2019-11-20:
I've made some boards with the 16 MBit chips: W25Q128JV_SQ, W25Q128JV_PM as well as a new IS25LP128F, which is not included in flash_devices.h but is included in circuitpython devices.h. None of them work out of the box so people should be careful using these larger chips, which I unfortunately need for audio files. I've tested boards with these chips with both circuitpython as well as this library. The code hangs right here for the IS25LP128F, which is the only one that at least appears to be recognized properly through JEDEC ID:

`  // The write in progress bit should be low.
  while ( readStatus() & 0x01 ) {}
	Serial.println("YOU SEE THIS ...");
  // The suspended write/erase bit should be low.
  while ( readStatus2() & 0x80 ) {}
	Serial.println("BUT DON'T SEE THIS ...");`

hathach, 2019-11-21:
since we don't have such as device and resource, we won't be able to test/fix this.

Casey10110, 2019-11-21:
I'd recommend taking these boards off the list of devices ASAP to prevent further loss of money to those using circuitpython in products. Having personally lost a good deal of money on this, I believe that only chips that work should be on there. I'd be surprised if anyone would disagree with this opinion.

hathach, 2019-11-21:
`flash_devices` is mirrored from cpython, I won't bother modifying it at all. It may need an extra command or timing etc... to works. We don't have your hardware, it could be possibly the wiring/pcb 
 or hw issue other than pure software driver. Maybe you could try to put a bit more effort to troubleshoot/analyze its issues like how is the QSPI command & response on the logic analyzer etc ... We know it doesn't work, but we need to also know why it doesn't work, right !!

Casey10110, 2019-11-21:
I completely understand and I can't tell you how much I appreciate all the work you guys do. I can't really figure it out and am going to put that design on the back-burner for now. I'm pretty sure it is software-related because I have another smaller chip working fine on the exact same board. Just too difficult to pinpoint the issue at this point. I wouldn't be surprise if somebody struggling with the exact same issues comes across my posts here. Thank you again for all your work.

hathach, 2020-03-11:
closed due to lack of activities, can be reopened if there is people willing to investigate further into the issue.

closed 2020-03-11 by hathach