adafruit/Adafruit_SPIFlash #33: Don't waste memory if not using SdFat

mzero, 2019-12-01:
The class `Adafruit_SPIFlash` is a fine interface if you want to use SPI flash, but without the FAT file system.

However, it includes both virtual methods for the `BaseBlockDriver` API, and an instance variable of type `Adafruit_FlashCache`. The later uses 4k+ of RAM. If you are not using FAT, or the block interface directly - this is just wasted RAM.

I suggest splitting  the `Adafruit_SPIFlash` class like so:

```C++
class Adafruit_SPIFlashCore {
	Adafruit_SPIFlashCore(Adafruit_FlashTransport* transport);
	~Adafruit_SPIFlashCore() {}

  ... most of what is in Adafruit_SPIFlash today ...
};

class Adafruit_SPIFlash : public Adafruit_SPIFlash {
	Adafruit_SPIFlash(Adafruit_FlashTransport* transport)
		: Adafruit_SPIFlashCore(transport) {}
	~Adafruit_SPIFlash() {}

	//------------- SdFat BaseBlockDRiver API -------------//
	virtual bool readBlock(uint32_t block, uint8_t* dst);
	virtual bool syncBlocks();
	virtual bool writeBlock(uint32_t block, const uint8_t* src);
	virtual bool readBlocks(uint32_t block, uint8_t* dst, size_t nb);
	virtual bool writeBlocks(uint32_t block, const uint8_t* src, size_t nb);

private:
	Adafruit_FlashCache _cache;
};
```
`Adafruit_SPIFlashCore` should be in its own header file, which doesn't `#include SdFat.h` - so that it works on systems that didn't load that library.

ladyada, 2019-12-01:
@hathach will look when back from holiday

mzero, 2019-12-03:
A basic proof of concept can be seen in this commit: 81048fece9ea1017e31be47a1a3c8f9a2d2c2493

If the sketch uses `Adafruit_SPIFlashCore` instead of `Adafruit_SPIFlash`, it'll save 4104 bytes of static data in SRAM, and about 500 or so bytes of program memory.

I'm not wedded to that name by any means, but I wanted to keep `Adafruit_SPIFlash` the same as existing sketches will expect it to be the full featured version.

It would be nice to break this into two .h and two .cpp files so that SdFat library doesn't need to be loaded if you don't need it ... but there doesn't seem to be any way to get the IDE to not compile all the files in a used library.

hathach, 2019-12-09:
@mzero yeah, we can split the class as you suggested, would you mind submitting an PR for this issue. I am happy to review and merge it.

mzero, 2019-12-09:
Sure thing. If you are planning on pulling my other change, I'll keep this one on top of it. Is the name `Adafruit_SPIFlashCore` good? Or perhaps `Adafruit_SPIFlashBase`? Or? I'll add some commentary explaining why it is there, too.

hathach, 2019-12-09:
@mzero `Adafruit_SPIFlashBase` makes more sense to me. Thank you. I am looking forward to  your PR.

PR #41 (2019-12-11): split Adafruit_SPIFlash into two classes
- saves 4104 bytes of Static Data in SRAM if you don't
 need the BaseBlockDriver functions!

- new class Adafruit_SPIFlashBase
- new header & src files Adafruit_SPIFlashBase.h/.cpp
- moved SFLASH_CMD constants into Adafruit_FlashTransport.h
  - this way the transport implementations need only import that
- made SPIFLASH_LOG in Adafruit_SPIFlashBase.cpp write in terms
  of addresses, not LBAs, since its API isn't in terms of blocks

Fixes #33
closed 2019-12-11 by hathach