adafruit/Adafruit_SPIFlash #34: writeBuffer fails with non-page aligned addresses

mzero, 2019-12-02:
The function `Adafruit_SPIFlash::writeBuffer` takes an address in flash, a pointer to data, and a length. It splits up the write into page sized operations (the largest the underlying chips can program at a time), and continues until the whole buffer is written.

The code correctly handles writes smaller than a page, and writes where the last chunk is less than a full page. Flash chips correctly handle these partial page program operations, leaving the other parts of the page intact.

However, the code in `Adafruit_SPIFlash::writeBuffer` incorrectly handles the case where the address is not page aligned, _and_ the length is long enough to spill into the next page.

For example:

```C++
    flash.writeBuffer(0x0020, mydatabuffer, 320);
```

In this case, the current code will program two chunks:
* @ address 0x0020, for 256 bytes
* @ address 0x0120, for 64 bytes

Flash chips do not support the first operation in the way one might expect: They _wrap the data at the page boundary,_ and so bytes 224 to 255 are written into addresses 0x0000 to 0x0019, that is, on the first page, in front of bytes 0 to 223!

The correct chunking should be:
* @ address 0x0020, for 224 bytes
* @ address 0x0100, for 96 bytes

The following function correctly computes the chunking: 

```C++
  uint32_t writeBuffer (
    Adafruit_SPIFlash& flash,
    uint32_t address, uint8_t const *buffer, uint32_t len)
  {
    uint32_t remain = len;

    // write one page at a time
    while(remain)
    {
      uint32_t const leftOnPage
        = SFLASH_PAGE_SIZE - (address & (SFLASH_PAGE_SIZE - 1));

      uint32_t const toWrite = min(remain, leftOnPage);

      if (flash.writeBuffer(address, buffer, toWrite) != toWrite) break;

      remain -= toWrite;
      buffer += toWrite;
      address += toWrite;
    }

    return len - remain;
  }
```

The logic of `leftOnPage` should be incorporated into `Adafruit_SPIFlash::writeBuffer`.

ladyada, 2019-12-02:
ooh wanna submit a PR? @hathach can review when back from holiday

mzero, 2019-12-02:
yup - will do in a day or two - just need to get my other code squared away!

PR #36 (2019-12-09): Fix bug in writeBuffer
fixes #34
closed 2019-12-09 by hathach