adafruit/Adafruit_SPIFlash #38: Conflict with ESP8266 core

caternuson, 2019-12-09:
- ESP8266 BSP = 2.6.2
- Target Board = "Adafruit Feather HUZZAH ESP8266"

Test sketch:
```cpp
#include <Adafruit_SPIFlash.h>
#include "ESP8266WiFi.h"

void setup() {
}

void loop() {
}
```

error:
```
/home/user/.arduino15/packages/esp8266/hardware/esp8266/2.6.2/cores/esp8266/FS.h:258:11: error: 'File' is already declared in this scope
 using fs::File;
```

ladyada, 2019-12-09:
yep there's no generic File type, so this happens. would be fixed by 
https://github.com/adafruit/Adafruit_SPIFlash/issues/33

mzero, 2019-12-11:
If @caternuson is including `Adafruit_SPIFlash.h` because they intend to use it like so:

```C++
#include <Adafruit_SPIFlash.h>

Adafruit_SPIFlash flash(&flashTransport);
FatFileSystem fatfs;

void setup() {
  ...
  flash.begin();
  fatfs.begin(&flash);
  ...
}
```
Then even after the split proposed in #33 there will still be a conflict:
 * `Adafruit_SPIFlash.h`
 * includes `SdFat.h`
 * includes `FatLib/FatLib.h`
 * includes `FatLib/ArduinoFiles.h`
 * which declares `class File` in global scope

On the other hand, the ESP8266 board package includes a core file `FS.h`, which also declares `class File` in global scope. There wouldn't be a conflict as long as you don't try to use both.... but alas, many of the ESP8266 libraries _do_ include its `FS.h`... so if your sketch is using _those_, then you get the conflict.

I don't know enough about the ESP8266 boards to do know if it would ever make sense to have both its file system in the same sketch with a separate SPI FLash file system... seems unlikely, but possible.

## On the other hand

If @caternuson is _not_ trying to put an `SdFat` file system on the SPI Flash, but only use it directly.... then after the split proposed in #33, things will be fine, as they'll just do:

```C++
#include <Adafruit_SPIFlashBase.h>

Adafruit_SPIFlashBase flash(&flashTransport);

void setup() {
  ...
  flash.begin();
  ...
  // use flash object directly, reading, erasing and writing as needed
}
```

caternuson, 2019-12-11:
@mzero I'm just proxying for someone else. See support forum thread linked above. I have no specific use case.

mzero, 2019-12-11:
I read through the forum post. They are indeed wanting to use `SdFat` on an SPI Flash memory - while also using WiFi features of the ESP8266 board. Saddly, this means that both `File` classes need to be used, but both are in global scope, and so conflict.

Possible fixes:
1) Rename one or the other `File` classes.
2) Wrap a namespace around `SdFat`'s globals.

Of course, both of these would break existing sketches.

The right long term solution is probably a comprehensive use of namespaces for libraries, using some consistent, predictable scheme, so that upgrading is easy - (no more difficult than adding a `use` statement for older code.)

hathach, 2022-10-17:
this should be fixed with latest release 4.0.0 . Where SdFat is upgraded to v2 and File is not defined for esp8266 core (and rp2040 as well). For the best compatible File32 should be used

closed 2022-10-17 by hathach