adafruit/Adafruit_SPIFlash #39: Updated to version 3.1.4

GonzOne, 2019-12-10:
I have a small project that was working on a Adafruit Feather HUZZAH ESP8266, I updated my libraries and the project no longer compiles. I stripped away the code to only debug the essential libraries and the following still does not compile.

#include <Adafruit_GFX.h>         // Core graphics library
#include <Adafruit_HX8357.h>      // Hardware-specific library
#include <SdFat.h>                // SD card & FAT filesystem library
#include <Adafruit_SPIFlash.h>    // SPI / QSPI flash library

  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}

Im getting this error;

In file included from /Users/manuelgonzalez/Documents/Arduino/debug_sketch/debug_sketch.ino:8:0:
/Users/manuelgonzalez/Documents/Arduino/libraries/Adafruit_SPIFlash/src/Adafruit_SPIFlash.h:39:4: error: #error ENABLE_EXTENDED_TRANSFER_CLASS must be set to 1 in SdFat SdFatConfig.h
   #error ENABLE_EXTENDED_TRANSFER_CLASS must be set to 1 in SdFat SdFatConfig.h
    ^
/Users/manuelgonzalez/Documents/Arduino/libraries/Adafruit_SPIFlash/src/Adafruit_SPIFlash.h:43:4: error: #error FAT12_SUPPORT must be set to 1 in SdFat SdFatConfig.h
   #error FAT12_SUPPORT must be set to 1 in SdFat SdFatConfig.h
    ^
exit status 1
Error compiling for board Adafruit Feather HUZZAH ESP8266.

How do I resolve this issue? Thanks!

ladyada, 2019-12-10:
try doing the things it is asking you to do

GonzOne, 2019-12-11:
Thanks Ladyada, It seems I have a host of other issues that i need to debug. Thanks for your quick response!

closed 2019-12-11 by GonzOne