adafruit/Adafruit_SPIFlash #44: Can't format SPI flash on Feather M0 Express

jakubhlavaty, 2020-02-09:
I want to format the onboard flash on the Feather M0 Express and store some files there.
The format example gives the following output:

```
Adafruit SPI Flash FatFs Format Example
Flash chip JEDEC ID: 0x14015
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
This sketch will ERASE ALL DATA on the flash chip and format it with a new filesystem!
Type OK (all caps) and press enter to continue.
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
Creating and formatting FAT filesystem (this takes ~60 seconds)...
Formatted flash!
Error, failed to mount newly formatted filesystem!

```

ladyada, 2020-02-09:
please make sure you have the latest SAMD package, we made some changes recently! :)

hathach, 2020-02-10:
@jakubhlavaty I just pull out my Feather M0 Express to test, seem to be good with latest lib version, can you confirm your lib version is at least as follows (it is part of compile verbose output)

```
Using library SPI at version 1.0 in folder: 
Using library Adafruit_ZeroDMA at version 1.0.4 in folder: 
Using library SdFat at version 1.2.1 in folder: 
Using library Adafruit_SPIFlash at version 3.1.4 in folder:
```

If it still fails, please try to eanble SdFat DEBUG_MODE to 1 and re-compile & upload, then post your Serial output here https://github.com/adafruit/SdFat/blob/master/src/FatLib/FatVolume.h#L38

hathach, 2020-02-10:
PS: I notice your board using a different flash device than mine, your seem to be `S25FL116K`, mine is `GD25Q16C` . But the lib should technically work with both. 

```
Adafruit SPI Flash FatFs Format Example
Flash chip JEDEC ID: 0xC84015
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
This sketch will ERASE ALL DATA on the flash chip and format it with a new filesystem!
Type OK (all caps) and press enter to continue.
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
Creating and formatting FAT filesystem (this takes ~60 seconds)...
Formatted flash!
Flash chip successfully formatted with new empty filesystem!
```

jakubhlavaty, 2020-02-10:
Just a hint: I had some issues compiling so I downgraded the boards lib to the pre-last version which solved that issue. Not sure if it could be related.

I will try the debug option, thanx :-)

jakubhlavaty, 2020-02-10:
Btw this is on windows. I can try on Linux as well.

hathach, 2020-02-11:
> Just a hint: I had some issues compiling so I downgraded the boards lib to the pre-last version which solved that issue. Not sure if it could be related.
> 
> I will try the debug option, thanx :-)

Thanks for your info, but please be specific about your SAMD BSP, SPIFlash and SdFat (Adafruit's Fork) version that you are using. Which version that works and which doesn't. Yes if possible the debug log will help to analyze the problem.

jakubhlavaty, 2020-02-11:
- Arduino SAMD: 1.8.2
- Adafruit SAMD: 1.5.9 (1.5.10 was the one that I had trouble compiling with and found the hint to downgrade online)
- Adafruit circuit playground: 1.10.4
- Adafruit SPIFlash: 3.1.4
- SdFat - Adafruit fork: 1.2.3

also some other Adafruit libs:
- Adafruit SSD1306: 2.1.0
- Adafruit STMPE610: 1.1.0
- Adafruit VS1053: 1.0.8
- Adafruit GFX: 1.7.5
- Adafruit ILI9341: 1.5.4

I didn't yet have the chance to run with debug. I'll let you know when I do.

jakubhlavaty, 2020-02-25:
Just tried it on Linux and everything works like a charm.
Fedora 26  4.16.11-100.fc26.x86_64

I had an older version of Arduino IDE lying around so I installed the latest one (1.8.11) and added the new SPIFlash libs.

Adafruit SAMD: 1.0.21
Adafruit circuit playground: 1.7.0
Adafruit SPIFlash: 3.1.4
SdFat - Adafruit fork: 1.2.3

I guess you can just close this one :-)

Btw I was impressed by your quick reply and btw2 just want to say you're awesome @ladyada.

Have a great day.

hathach, 2020-02-27:
thanks for confirming the issue

closed 2020-02-27 by hathach