adafruit/Adafruit_SPIFlash #46: Support for overriding flash type autodetection

modmatic, 2020-03-04:
It would be nice to be able to specify an explicit flash device, rather than rely on auto-detection. Currently, it seems that not all devices in `flash_devices.h` are detectable, due to not existing in `possible_devices` (`Adafruit_SPIFlashBase.cpp`). Perhaps there could be some simple API for passing in some explicit device (likely from `flash_devices.h`), such as in the `begin()` method. Thoughts?

It would be better to point users to your library, rather than maintain a fork.

Great work otherwise! Thanks for the awesome library :)

modmatic, 2020-03-04:
For the record, I'm trying to use `W25Q16JV_IQ`. Will have to fork for now and insert it.

hathach, 2020-03-05:
@modmatic that is totally possible, update begin() to take external_flash_device* parameter, default to NULL for auto detection. Else it will use the input devices. If you could submit an PR, I will be glad to merge it.

modmatic, 2020-03-05:
@hathach Great! I'd be happy to contribute a PR

(Now I just need to get `W25Q16JV_IQ` to actually work with any of the examples, especially the format example. But that's a separate issue.)

PR #47 (2020-03-11): Add ability to provide explicit device in begin()
Resolves #46

Non-breaking change
commit 66e9ced2 (2020-03-11): Add ability to provide explicit device in begin()

Resolves #46
PR #50 (2020-03-21): allow to provide flash device in begin()
fix #46
closed 2020-03-21 by hathach