adafruit/Adafruit_SPIFlash #51: Support to SST26VF064B Flash

vslinuxdotnet, 2020-04-11:
Hello,

I try to add support for SST26VF064B with
```
#define SST_SST26VF064B {\
    .total_size = (1 << 23), /* 8 MiB */ \
    .start_up_time_us = 10000, \
    .manufacturer_id = 0xbf, \
    .memory_type = 0x26, \
    .capacity = 0x43, \
    .max_clock_speed_mhz = 104, \
    .quad_enable_bit_mask = 0x01, \
    .has_sector_protection = false, \
    .supports_fast_read = true, \
    .supports_qspi = true, \
    .supports_qspi_writes = true, \
    .write_status_register_split = false, \
    .single_status_byte = false, \
}
```
I add the define in: static const external_flash_device possible_devices[]

All seams ok but the MCU freeze in detection.
Any parameter in the flash define is incorrect ?

I the file Adafruit_SPIFlash.cpp if none flash was detect I suggest to change from:
`if (_flash_dev == NULL) return false;`
to
`if (_flash_dev == NULL || jedec_ids[0] == 255 ) return false;`

Because the flash begin() init always return true if no device detected.

Thank you

hathach, 2020-04-13:
we cannot help you on this as we have no flash device to test with. If you could manage to get it work, please consider to submit a PR to help others with the same problem.

mzero, 2020-06-05:
@vslinuxdotnet - You suggested change seems to suggest a different issue: Is your concern that if there is no flash device at all, that `jedec_id[0]` is 255, and that should fail? The current behavior is that if you supply a pointer to a device, the library will use that no matter what. Are you suggesting that it should return `false` in this case when the detected `jedec_id` doesn't match the supplied device?

hathach, 2020-07-14:
closed due to lack of activities.

closed 2020-07-14 by hathach