adafruit/Adafruit_SPIFlash #56: File System corruption with 32MB flash after writing ~16MB

jacobmcollier, 2020-06-23:
I am using W25Q256JV to save wav files. I have it up and running using sdfat and this library and can save files to the flash and read them back out. However once I save enough files to the flash to exceed about half the capacity of the flash (~16MB) the file system becomes corrupt and will no longer mount.

Was this library written for a max flash size of 16MB? Or is this just a potential bug? Any ideas on where to start digging to save me some time would be appreciated!

hathach, 2020-06-23:
hmm, the spi flash use FAT12 for vast majority of the flash device
https://en.wikipedia.org/wiki/File_Allocation_Table#FAT12

And look like the limit size of FAT12 is 16MB with default 4KB cluster. So yeah, I think you are hitting the limit. It should be considered as a **bug**. I am currently busy with other works atm, I will try to make it more dynamically to use FAT16 for larger device when possible.

PS: since I don't have your device, I will ping you to test with when the PR is ready.

jacobmcollier, 2020-06-25:
Thanks for the response and this great library!

I'd be happy to test out the update. Any guess when you might get to this??? Two days? Two weeks? Two months? I'm just trying to figure out the time frame for my current project.

jacobmcollier, 2020-07-06:
I've done just a little more digging into this issue and printed out some of the parameters for the FATFS object after formatting:
```
SPI Flash CLI Interface
- Flash chip JEDEC ID: 0xEF4019
> format
Creating and formatting FAT filesystem...
Complete!!!
Info:
fs_type:   2
pdrv:      0
n_fats:    1
n_rootdir: 512
csize:     8
last_clst: 4294967295
n_fatent:  8185
fsize:     39
```

According to the documentation fs_type = 2 indicates it's formatted for FAT16. Thoughts???

I have a bit of time I can dig in more this week.

jacobmcollier, 2020-08-26:
I've identified the root of the issue I was facing. To utilize the full memory space of the 256Mb flash you need to use the 4-byte addressing mode. This library was written to utilize the 3-byte address mode for smaller chips. To prove the issue I modified
`Adafruit_FlashTransport_SPI::eraseCommand`
`Adafruit_FlashTransport_SPI::readMemory`
`Adafruit_FlashTransport_SPI::writeMemory`
to use a 32-bit address and added the command to enter 4-byte address mode in
`Adafruit_SPIFlashBase::begin`

Also I updated
`Adafruit_SPIFlashBase::numPages`
to return a uint32_t instead of uint16_t to avoid a rollover error.

A third change was a simple optimization to increase throughput by more than 2x. In
`Adafruit_FlashTransport_SPI::readMemory`
I replaced the while loop with
`_spi->transfer(data, len);`

hathach, 2020-08-26:
> I've identified the root of the issue I was facing. To utilize the full memory space of the 256Mb flash you need to use the 4-byte addressing mode. This library was written to utilize the 3-byte [...]

ah that is indeed an root cause, I don't have the chip to test with, would you mind submitting the PR for 32-bit address for supported device. Please make sure it should still work with existing 24-bit address as it is.  

> A third change was a simple optimization to increase throughput by more than 2x. In
> `Adafruit_FlashTransport_SPI::readMemory`
> I replaced the while loop with
> `_spi->transfer(data, len);`

You should update from the master first, there is a great improvement for reading/writing that I have just committed recently.

hathach, 2020-08-26:
@jacobmcollier you are really in luck, it just happens that one of our fram flash is too small (8KB) and only use 16-bit address. Therefore I have updated the code to support multiple address width here https://github.com/adafruit/Adafruit_SPIFlash/pull/70 . Let's me know that works for you, also which board and MCU you are using ?

jacobmcollier, 2020-08-29:
I was able to test out the updates from #70 and I had to make some additional changes for it to compile and function properly for my setup. See #71. My setup consists of a custom board with the STM32L4 MCU.

hathach, 2020-08-31:
closed by #71

closed 2020-08-31 by hathach