adafruit/Adafruit_SPIFlash #57: Adapting Library for different chip

WJRabieSAT, 2020-07-02:
Hi. I would like some help with using this library on the adafruit feather M0 adalogger and S25FL128L flash chip.

I am trying to get the msc_external_flash_sdcard example to work on an adafruit feather M0 adalogger with the flash chip SCK on pin 13, MISO on pin 12, MOSI on pin 11 and CS on pin 6.

The SD card is working perfect. I can use it over usb and in code perfectly. Now I want to get the flash working.

I created the new SPI Class for the pins I am using.

SPIClass flashSPI (&sercom1, 12, 13, 11, SPI_PAD_0_SCK_1, SERCOM_RX_PAD_3);
#define flash_cs  6   //Chip select pin
Adafruit_FlashTransport_SPI flashTransport(flash_cs, flashSPI);
Adafruit_SPIFlash flash(&flashTransport);
FatFileSystem fatfs;

When I read JEDEC ID, I get a value of 00d5h.
When I read page size I get 256.
When I read number of pages I get 0.

With the following:

    usb_msc.setMaxLun(2);
    usb_msc.setID(0, "Flash Memory", "External Flash", "1.0");
    usb_msc.setID(1, "SD Memory", "SD Card", "1.0");
    usb_msc.begin();

"SD Memory" and "Flash Memory" are both detected as storage devices over USB in task manager. The SD works perfectly but not the Flash.

Any help will be appreciated.

Thank you.

hathach, 2020-07-03:
please try with https://github.com/adafruit/Adafruit_SPIFlash/tree/master/examples/flash_info and post your serial output. If you modify the example. post the whole sketch here as well.

WJRabieSAT, 2020-07-03:
Hi. Thanks for the reply.

Since I am using the adalogger M0 and not the M0 Express, there is no QSPI/SPI flash defined in variant.h.

So I have to add it to the project.

I can read and write to the flash when I use a different library, but then I dont have the file system and usb options that I need.

I will post the code I used for flash_info.ino and the serial output.

WJRabieSAT, 2020-07-03:
// The MIT License (MIT)
// Copyright (c) 2019 Ha Thach for Adafruit Industries

#include "SdFat.h"
#include "Adafruit_SPIFlash.h"
#include "wiring_private.h"

// On-board external flash (QSPI or SPI) macros should already
// defined in your board variant if supported
// - EXTERNAL_FLASH_USE_QSPI
// - EXTERNAL_FLASH_USE_CS/EXTERNAL_FLASH_USE_SPI

/*
#if defined(EXTERNAL_FLASH_USE_QSPI)
  Adafruit_FlashTransport_QSPI flashTransport;

#elif defined(EXTERNAL_FLASH_USE_SPI)
  Adafruit_FlashTransport_SPI flashTransport(EXTERNAL_FLASH_USE_CS, EXTERNAL_FLASH_USE_SPI);

#else
  #error No QSPI/SPI flash are defined on your board variant.h !
#endif
*/

SPIClass flashSPI (&sercom1, 12, 13, 11, SPI_PAD_0_SCK_1, SERCOM_RX_PAD_3);
#define flash_cs  6   //Chip select pin

Adafruit_FlashTransport_SPI flashTransport(flash_cs, flashSPI);

Adafruit_SPIFlash flash(&flashTransport);

// the setup function runs once when you press reset or power the board
void setup()
{
  Serial.begin(115200);
  while ( !Serial ) delay(10);   // wait for native usb
  
  pinMode(flash_cs, OUTPUT);
  
  flash.begin();
  pinPeripheral(11, PIO_SERCOM);
  pinPeripheral(12, PIO_SERCOM);
  pinPeripheral(13, PIO_SERCOM);
  
  Serial.println("Adafruit Serial Flash Info example");
  Serial.print("JEDEC ID: "); Serial.println(flash.getJEDECID(), HEX);
  Serial.print("Flash size: "); Serial.println(flash.size());
}

void loop()
{
  // nothing to do
}

WJRabieSAT, 2020-07-03:
Adafruit Serial Flash Info example
JEDEC ID: D5
Flash size: 0

hathach, 2020-07-03:
please wrap both code and output in the block quote

WJRabieSAT, 2020-07-03:
When I use the following library: https://github.com/Marzogh/SPIMemory

I can read and write to the flash and I get the following info from the flash:

Jedec BIN: 10110000000011000
Jedec DEC: 90136
Jedec HEX: 16018
ManID DEC: 0
Manufacturer ID: 01h
Memory Type: 60h
Capacity: 18h
JEDEC ID: 16018h

hathach, 2020-07-08:
By default, the libraries will only know to work with the list of common flash devices, mainly the one used in Adafruit's products
https://github.com/adafruit/Adafruit_SPIFlash/blob/master/src/Adafruit_SPIFlashBase.cpp#L25
It is matched using the JEDEC ID

Since your flash device is not part of default list, you need to pass it as additional flash device into `begin(flash_devs, count)` variant https://github.com/adafruit/Adafruit_SPIFlash/blob/master/src/Adafruit_SPIFlashBase.h#L50

I did a quick check, you are out of luck since the big supported list doesn't have your device, but don't worry it can be simply construct using the chip manual. You could take a look at how these are defined here.
https://github.com/adafruit/Adafruit_SPIFlash/blob/master/src/flash_devices.h

WJRabieSAT, 2020-07-23:
Hi. Sorry for the late reply. This was on hold for a while but I am busy with it again. 

Do you have an example of how "begin(flash_devs, count)" is to be used?

hathach, 2020-07-23:
The API for custom device is already implemented. I close this now since it is not a bug nor feature. Please post an topic in support forum. There are a lot more user base there to help you out with how-to.

closed 2020-07-23 by hathach