adafruit/Adafruit_SPIFlash #58: Formatting SPI flash doesn't create a Volume

AlexThePotter, 2020-07-06:
Hi, I'm using a Feather M0 adalogger, with an external W25Q64.

I have it wired to Hardware SPI and CS = A1.

Using the example, I first run flash_erase which succeeds.

Then when I run SdFat_format I get the following output:

`Adafruit SPI Flash FatFs Format Example
Flash chip JEDEC ID: 0xEF4017
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
This sketch will ERASE ALL DATA on the flash chip and format it with a new filesystem!
Type OK (all caps) and press enter to continue.
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
Creating and formatting FAT filesystem (this takes ~60 seconds)...
Error, f_mount failed with error code: 13
`

Error 13, meaning no volume exists, but the previous command to create the file system passes without issue.

I'm running the examples as stock except for changing this section:

`Adafruit_FlashTransport_SPI flashTransport(A1,SPI);
Adafruit_SPIFlash flash(&flashTransport);`

hathach, 2020-07-14:
my comment on other issue may be helpful 

By default, the libraries will only know to work with the list of common flash devices, mainly the one used in Adafruit's products
https://github.com/adafruit/Adafruit_SPIFlash/blob/master/src/Adafruit_SPIFlashBase.cpp#L25
It is matched using the JEDEC ID

Since your flash device is not part of default list, you need to pass it as additional flash device into `begin(flash_devs, count)` variant https://github.com/adafruit/Adafruit_SPIFlash/blob/master/src/Adafruit_SPIFlashBase.h#L50

I did a quick check, you are out of luck since the big supported list doesn't have your device, but don't worry it can be simply construct using the chip manual. You could take a look at how these are defined here.
https://github.com/adafruit/Adafruit_SPIFlash/blob/master/src/flash_devices.h

PS: Next time, please put all the code and serial output into the block quote for readability

_Originally posted by @hathach in https://github.com/adafruit/Adafruit_SPIFlash/issues/57#issuecomment-655626931_

hathach, 2020-07-14:
please try with https://github.com/adafruit/Adafruit_SPIFlash/tree/master/examples/flash_info and post your serial output. If you modify the example. post the whole sketch here as well.

hathach, 2020-08-03:
closed due to lack of activities

igortomiatti, 2021-07-05:
I'm having the same problem with the W25Q64.

I checked the flash_devices file and it now has support for this flash.