adafruit/Adafruit_SPIFlash #61: Nano BLE and MbedOS Arduino error

sabas1080, 2020-07-10:
Hi I am test this library with Arduino Nano BLE and Arduino Core Mbed

But I have the error

```
/Users/xxx/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -c -w -g -Os -nostdlib @/Users/xx/Library/Arduino15/packages/arduino/hardware/mbed/1.1.4/variants/bast_ble/defines.txt @/Users/xxx/Library/Arduino15/packages/arduino/hardware/mbed/1.1.4/variants/bast_ble/cxxflags.txt -DARDUINO_ARCH_NRF52840 -MMD -mcpu=cortex-m4 -DARDUINO=10810 -DARDUINO_bast_ble -DARDUINO_ARCH_MBED -I/Users/xxx/Library/Arduino15/packages/arduino/hardware/mbed/1.1.4/cores/arduino -I/Users/xxx/Library/Arduino15/packages/arduino/hardware/mbed/1.1.4/variants/bast_ble -I/Users/xxx/Documents/Arduino/libraries/SdFat_-_Adafruit_Fork/src -I/Users/xxx/Library/Arduino15/packages/arduino/hardware/mbed/1.1.4/libraries/SPI -I/Users/xxx/Documents/Arduino/libraries/Adafruit_SPIFlash/src -I/Users/xxx/Library/Arduino15/packages/arduino/hardware/mbed/1.1.4/cores/arduino/api/deprecated -iprefix/Users/xxx/Library/Arduino15/packages/arduino/hardware/mbed/1.1.4/cores/arduino @/Users/xxx/Library/Arduino15/packages/arduino/hardware/mbed/1.1.4/variants/bast_ble/includes.txt /var/folders/mv/_ygj1fx50f9_sd6zn6fgyt3h0000gn/T/arduino_build_236814/sketch/flash_info.ino.cpp -o /var/folders/mv/_ygj1fx50f9_sd6zn6fgyt3h0000gn/T/arduino_build_236814/sketch/flash_info.ino.cpp.o
In file included from /Users/xxx/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/arm-none-eabi/include/sys/_default_fcntl.h:200:0,
                 from /Users/xxxx/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/arm-none-eabi/include/sys/fcntl.h:4,
                 from /Users/xxx/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/arm-none-eabi/include/fcntl.h:1,
                 from /Users/xxx/Documents/Arduino/libraries/SdFat_-_Adafruit_Fork/src/FatLib/FatApiConstants.h:30,
                 from /Users/xxx/Documents/Arduino/libraries/SdFat_-_Adafruit_Fork/src/FatLib/FatFile.h:36,
                 from /Users/xxx/Documents/Arduino/libraries/SdFat_-_Adafruit_Fork/src/FatLib/ArduinoFiles.h:33,
                 from /Users/xxx/Documents/Arduino/libraries/SdFat_-_Adafruit_Fork/src/FatLib/FatLib.h:27,
                 from /Users/xxxx/Documents/Arduino/libraries/SdFat_-_Adafruit_Fork/src/SdFat.h:33,
                 from /Users/xxxx/Documents/Arduino/libraries/Adafruit_SPIFlash/examples/flash_info/flash_info.ino:4:
/Users/xxxxx/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/arm-none-eabi/include/sys/stat.h:27:8: error: redefinition of 'struct stat'
 struct stat
        ^~~~
In file included from /Users/xxx/Library/Arduino15/packages/arduino/hardware/mbed/1.1.4/cores/arduino/mbed/platform/mbed_error.h:21:0,
                 from /Users/xxxx/Library/Arduino15/packages/arduino/hardware/mbed/1.1.4/cores/arduino/mbed/rtos/Queue.h:28,
                 from /Users/xxx/Library/Arduino15/packages/arduino/hardware/mbed/1.1.4/cores/arduino/mbed/rtos/Mail.h:28,
                 from /Users/xxx/Library/Arduino15/packages/arduino/hardware/mbed/1.1.4/cores/arduino/mbed/rtos/rtos.h:33,
                 from /Users/xxx/Library/Arduino15/packages/arduino/hardware/mbed/1.1.4/cores/arduino/mbed/mbed.h:23,
                 from /Users/xxxx/Library/Arduino15/packages/arduino/hardware/mbed/1.1.4/cores/arduino/Arduino.h:32,
                 from /var/folders/mv/_ygj1fx50f9_sd6zn6fgyt3h0000gn/T/arduino_build_236814/sketch/flash_info.ino.cpp:1:
/Users/xxxxx/Library/Arduino15/packages/arduino/hardware/mbed/1.1.4/cores/arduino/mbed/platform/mbed_retarget.h:500:8: note: previous definition of 'struct stat'
 struct stat {
        ^~~~
Se encontraron varias bibliotecas para "Adafruit_SPIFlash.h"
Usado: /Users/xxxx/Documents/Arduino/libraries/Adafruit_SPIFlash
Se encontraron varias bibliotecas para "SdFat.h"
Usado: /Users/xxxx/Documents/Arduino/libraries/SdFat_-_Adafruit_Fork
Se encontraron varias bibliotecas para "SPI.h"
Usado: /Users/xxxxx/Library/Arduino15/packages/arduino/hardware/mbed/1.1.4/libraries/SPI
```

hathach, 2020-07-11:
Yeah, we didn’t test it with Arduino nano ble 33. If you could fix tye build please submit a PR to help others

sabas1080, 2020-07-12:
seems like a core bug
https://github.com/arduino/ArduinoCore-nRF528x-mbedos/issues/49

Solution: 
https://github.com/ARMmbed/mbed-os/pull/12607

Thnaks @hathach

closed 2020-07-12 by sabas1080