adafruit/Adafruit_SPIFlash #69: Why is the NeoPixel library included in the dependencies?

dojyorin, 2020-08-26:
NeoPixel is included in "depends" of library.properties.

NeoPixel may depend on SPIFlash, but I don't think SPIFlash depends on NeoPixel.
(I think NeoPixel is high-level and SPIFlash is low-level)

Please let me know if you need to depend on NeoPixel.
If you don't need a dependency, would you like to remove it from "depends"?
(I think simple is preferable for low-level)

hathach, 2020-08-26:
because one of the example needs it to build. The library does not use that. Writing your own sketch, you could just ignore it. and we won't change this dependency.

closed 2020-08-26 by hathach

dojyorin, 2020-08-27:
I understood that there is no dependency on the library itself.
Thank you for answer!

hathach, 2020-08-27:
> I understood that there is no dependency on the library itself.
> Thank you for answer!

Yeah, in sum up, lib itself is independent, but the deps is needed for our ci build script to compile all the example. Some of them do fancy thing like neopixel