adafruit/Adafruit_SPIFlash #72: Can't format flash - ItsyBitsy M4 Express with onboard flash

deezums, 2020-08-31:
For some reason I get an "Error, f_mount failed with error code: 13" when trying to run the SdFat_format example.

I am using the built in flash on the board, and it appears to be communicating correctly and responding 0xC84015 as it should. 

Am I doing something wrong, or is this a bug?

 - **PC & IDE** : Arduino 1.8.13 on Windows 7
 - **Board** : ItsyBitsy M4 Express
 - **BSP**   : Adafruit SAMD 1.6.0
 - **Sketch**: SdFat_format

**To Reproduce**
Steps to reproduce the behavior:
1. Load example to board, type "OK", hit enter...

```
Adafruit SPI Flash FatFs Format Example
Flash chip JEDEC ID: 0xC84015
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
This sketch will ERASE ALL DATA on the flash chip and format it with a new filesystem!
Type OK (all caps) and press enter to continue.
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
Creating and formatting FAT filesystem (this takes ~60 seconds)...
Error, f_mount failed with error code: 13

```

hathach, 2020-09-01:
There is numerous bug fix and improvement, could you try to use current tip of master branch for both SAMD BSP and SPIFLash to test again. If you don't know how to get the current latest code on git, just wait several days, we are rolling out the release for both of them.

deezums, 2020-09-01:
I tried using the BSP (1.6.1) and library from github but still get the same error. So I guess there's nothing I can do but wait now?

hathach, 2020-09-09:
can you try again with new BSP 1.6.2 and the latest SPIFlash 3.3.0 to see if it fixes your issue.

deezums, 2020-09-09:
I updated BSP and SPIFlash Libraries to 1.6.2 and 3.3.0, I still get the same error.

hathach, 2020-09-09:
hmm, I just pull out my Itsy M4 and is able to get it formatted without any issues. Your maybe corrupts somehow. Can you run the `flash_erase` first, which use an low level chip erase, then try again with the SD format.

```
Adafruit SPI Flash FatFs Format Example
Flash chip JEDEC ID: 0xC84015
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
This sketch will ERASE ALL DATA on the flash chip and format it with a new filesystem!
Type OK (all caps) and press enter to continue.
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
Creating and formatting FAT filesystem (this takes ~60 seconds)...
Setting disk label to: EXT FLASH
Formatted flash!
/home/hathach/Arduino/libraries/SdFat/src/FatLib/FatVolume.cpp478
Flash chip successfully formatted with new empty filesystem!

```

deezums, 2020-09-11:
I also have a Metro M4 and the code works fine on it, I guess I must have a hardware or deeper software issue on my end with my itsy M4. 

The flash erase example on the itsy M4 does work, but I get a confirmation almost instantly with no delay. The format still gives the same error. For curiosity's sake, I did the same test on my Metro M4 and it appears to take around 5 seconds before sending a confirmation response when erasing the chip, and the format is pretty quick but does work. 

I never tried to save data to the chip while using the very old 2.0.2 spiflash library, so I went back and ran the full example using version 2.0.2 and it works? So the hardware isn't bad.

I only have one itsy to test here, I believe I am using all the default board/compiler settings 120mHz, -Os, 50mHz qspi. 

Have I possibly missed a bootloader update or something?

hathach, 2020-09-11:
hmm, bootloader shouldn't be the issue, for now, I don't know what is your exact issue, can you try the `flash-speedtest` example
https://github.com/adafruit/Adafruit_SPIFlash/blob/master/examples/flash_speedtest/flash_speedtest.ino

It does erase, write, read and compare to test the speed. If there is error, please post your serial log here.

deezums, 2020-09-12:
```
Adafruit Serial Flash Speed Test example
JEDEC ID: C84015
Flash size: 2097152
Erase chip
Write flash with 0xAA
Write 2097152 bytes in 3.57 seconds.
Speed : 587.44 KB/s.

Read flash and compare
Error: flash contents mismatched at address 0x00000000!!!
000: EE EE EE EE EE EE EE EE EE EE EE EE EE EE EE EE 
...
FF0: EE EE EE EE EE EE EE EE EE EE EE EE EE EE EE EE
Erase chip
Write flash with 0x55
Write 2097152 bytes in 3.64 seconds.
Speed : 575.82 KB/s.

Read flash and compare
Error: flash contents mismatched at address 0x00000000!!!
000: 77 77 77 77 77 77 77 77 77 77 77 77 77 77 77 77 
...
FF0: 77 77 77 77 77 77 77 77 77 77 77 77 77 77 77 77
Speed test is completed.
```

Does this help at all?

hathach, 2020-09-13:
it helps a bit, the sketches try to 
- write 0xFF but read back with 0xEE
- write 0x55 but read back with 0x77

hmm, maybe the SPI clock is too fast on your setup (though It work fine with my Itsy M4). Can you try to manually edit SPIFlash max_clock_speed_mhz from 104 to 12 Mhz, if still not work, please try with 8MB. And post your serial output here.
 
https://github.com/adafruit/Adafruit_SPIFlash/blob/master/src/flash_devices.h#L96

deezums, 2020-09-13:
I still get the same errors, with speeds ranging from 560k at 104MHz to 445k at 4MHz, so I am pretty sure I am changing the proper line. 

0xFF was a typo, right? I thought the sketch tries 0xAA and 0x55, or alternating bits, and gets back an extra bit, but not in the same location.

This is the result for 12MHz

Adafruit Serial Flash Speed Test example
JEDEC ID: C84015
Flash size: 2097152
Erase chip
Write flash with 0xAA
Write 2097152 bytes in 3.96 seconds.
Speed : 529.85 KB/s.

Read flash and compare
Error: flash contents mismatched at address 0x00000000!!!
000: EE EE EE EE EE EE EE EE EE EE EE EE EE EE EE EE 

FF0: EE EE EE EE EE EE EE EE EE EE EE EE EE EE EE EE
Erase chip
Write flash with 0x55
Write 2097152 bytes in 3.97 seconds.
Speed : 527.72 KB/s.

Read flash and compare
Error: flash contents mismatched at address 0x00000000!!!
000: 77 77 77 77 77 77 77 77 77 77 77 77 77 77 77 77 

FF0: 77 77 77 77 77 77 77 77 77 77 77 77 77 77 77 77
Speed test is completed.

hathach, 2020-09-14:
> 0xFF was a typo, right? I thought the sketch tries 0xAA and 0x55, or alternating bits, and gets back an extra bit, but not in the same location.

Ah yeah, it is a typo, my brain auto map it to 0xFF when seeing 0xEE

> I still get the same errors, with speeds ranging from 560k at 104MHz to 445k at 4MHz, so I am pretty sure I am changing the proper line.
> 
> This is the result for 12MHz

hmm, can you take a picture of your board and upload it here, there are 3 revision of Itsybitsy M4, maybe your is a bit older with some hardware differences.

deezums, 2020-09-14:
Ordered jul 14th 2019, if that helps at all. 

[IMG_20200914_000448](https://user-images.githubusercontent.com/42391376/93046150-bc7d4700-f61e-11ea-8965-647d41abafaf.jpg)
[IMG_20200914_000439](https://user-images.githubusercontent.com/42391376/93046154-bedfa100-f61e-11ea-9b9e-33bcc0302b6f.jpg)

hathach, 2020-09-14:
thanks for the pix, it is revC, same version as mine. At this time, I would assume there is an issue with Quad mode, but single mode work well. Since we could read JEDEC from device correctly C84015 and you have no issue with v2 which only support single mode. What exactly is the root cause is not clear to me. 

Final check, did your IDE menu setup look the same to this especially with the CPU freq and QSPI max speed

[image]

deezums, 2020-09-16:
Sorry, I thought I had replied but I guess I didn't hit submit. 

I have been using the default CPU clock speed, QSPI speed, and compiler optimization if that were to matter. 120MHz, 50MHz, and -Os, in that order.

hathach, 2020-09-16:
sigh, to be honest, I have no clues what could go wrong here. Maybe it is a hardware issue or something else.

deezums, 2020-10-06:
I am pretty sure this one is a hardware issue. 

I have a metro m4 that works, itsy M0 that works, a few custom boards with the GD25, and a new one with a 16MB winbond chip. All of them work except my itsy M4. 

I did notice a few unused registers, ones that delay chip selects, clock pulses, and consecutive transfers. Testing with my other boards that will let me probe it appears these registers work as described. As a last ditch effort I tried copying some of these changes to the itsy m4 but it still doesn't work. 

Oh well, that's how it goes. Thanks for the awesome library!

closed 2020-10-06 by deezums