adafruit/Adafruit_SPIFlash #73: Adafruit_SPIFlashBase::end prototyped but not implemented.

wmcelderry, 2020-09-04:
**Describe the bug**
There is a prototype for 'Adafruit_SPIFlashBase::end', but it isn't implemented.
see Adafruit_SPIFlashBase.h:51 for the prototype, and notice the lack of implementation in corresponding files.

The reason I wanted it is that the NRF52 automatic clock generator could (depending on hardware implementation details I have not found documentation on) continue to generate clock signals until the underlying SPI peripheral in use is either stopped or suspended, which would increase the power usage in sleep mode generating a clock that is not really required.
The solution would be to stop the driver when the flash is not required for a while and start it before the next operation begins.

**Set up (mandatory)**
 - **PC & IDE** : N/A
 - **Board** : N/A
 - **BSP**   : N/A
 - **Sketch**: N/A

**To Reproduce**
Look at file Adafruit_SPIFlashBase.h and notice lack of corresponding implementation in  Adafruit_SPIFlashBase.cpp

hathach, 2020-09-04:
It is not a bug, it is a feature request, yeah, it isn't implemented just yet.

wmcelderry, 2020-09-08:
Fair enough!

I'd assumed 'bug' as the prototype exists and expected the implementation to be lost from refactoring. 

If I implement it, I will share the implementation.

I forgot to say: thanks for sharing your hard work!

hathach, 2020-09-09:
that is fair to say as well. Though the end() prototype is more a legacy from V2. Eventually we will implement it, just not now, since well we haven't needed it yet.

hathach, 2022-05-05:
implemented by #108