adafruit/Adafruit_SPIFlash #74: Unable to compile, missing declarations

mfarabee, 2020-09-20:
**Describe the bug**
I am compiling a program that uses Adafruit_ImageReader. The latest version requires Adafruit_SPIFlash. When compiling using Adafruit_SPIFlash version 3.2.2, everything compiles properly with no errors or warnings. When I upgrade to Adafruit_SPIFlash version 3.3.0, I receive many errors that state values are not declared in the scope (see below for details). When I revert back to 3.2.2, everything compiles correctly.  I wasx able to reproduce this in a very simple program (attached below).

**Set up (mandatory)**
 - **PC & IDE** : Arduino 1.8.12  Windows 10
 - **Board** : NodeMCU-32S
- **Libraries versions**
Adafruit BUSIO 1.50
Adafruit  EPD 2.53
Adafruit GFX 1.10.1
Adafruit ILI9341 1.56
Adafruit ImageReader 2.4.0
Adafruit SPIFlash 3.3.0
SdFat 1.1.4   <- with required changes 
                    //  SdFat library, editing the default SdFatConfig.h file
                   //       #define USE_STANDARD_SPI_LIBRARY 2           
                   //      #define FAT12_SUPPORT 1                       
                   //      #define ENABLE_EXTENDED_TRANSFER_CLASS 1     

 **To Reproduce**
Steps to reproduce the behavior:
Run the following code using the board NodeMCU-32S
============================================
#include <SdFat.h>
#include <SPI.h>
#include <Adafruit_GFX.h>       
#include <Adafruit_ILI9341.h> 
#include <Adafruit_ImageReader.h>
#include <Adafruit_ImageReader_EPD.h>
  Adafruit_ImageReader *reader;
  ImageReturnCode ImageReaderStat;
void setup() {
}
void loop() {
}
====================================

**Serial Log**
...\Arduino\libraries\Adafruit_SPIFlash\src/spi/Adafruit_FlashTransport_SPI.h:61:50: error: 'MSBFIRST' was not declared in this scope

     _spi->beginTransaction(SPISettings(clock_hz, MSBFIRST, SPI_MODE0));
Arduino\libraries\Adafruit_SPIFlash\src/spi/Adafruit_FlashTransport_SPI.h:62:23: error: 'LOW' was not declared in this scope

     digitalWrite(_ss, LOW);

                       ^
...\Arduino\libraries\Adafruit_SPIFlash\src/spi/Adafruit_FlashTransport_SPI.h:62:26: error: 'digitalWrite' was not declared in this scope

     digitalWrite(_ss, LOW);

                          ^
...\Arduino\libraries\Adafruit_SPIFlash\src/spi/Adafruit_FlashTransport_SPI.h: In member function 'void Adafruit_FlashTransport_SPI::endTransaction()':

...\Arduino\libraries\Adafruit_SPIFlash\src/spi/Adafruit_FlashTransport_SPI.h:66:23: error: 'HIGH' was not declared in this scope

     digitalWrite(_ss, HIGH);

                       ^
...\Arduino\libraries\Adafruit_SPIFlash\src/spi/Adafruit_FlashTransport_SPI.h:66:27: error: 'digitalWrite' was not declared in this scope

     digitalWrite(_ss, HIGH);

                           ^
...\Arduino\libraries\Adafruit_SPIFlash\src/Adafruit_SPIFlashBase.h:89:44: error: 'HIGH' was not declared in this scope

       digitalWrite(_ind_pin, _ind_active ? HIGH : LOW);

                                            ^
and many more ...........

hathach, 2020-09-21:
#75 should fix this issue, it would be great if you could test it out.

closed 2020-09-21 by hathach

mfarabee, 2020-09-21:
I was able to test it by compiling in the Arduino IDE version 1.8.13. It compiled with no errors. YEAH!

I was not able to upload the application to the ESP32 and test functionality, since I do not have access to my hardware until Wednesday, As soon as I have access to my hardware, I will test it and reply.

Thanks for the quick response!

hathach, 2020-09-21:
no problems, we will include esp32 as part of ci test in the next pr to make sure this won't occur in the future. Let's us know the result when you have the chance to test with your hardware.

mfarabee, 2020-09-23:
I was able to test this library using two programs on the ESP32. One used imageReader, SdFat and GFX libraries and it compiled and worked without a problem. This program read from the SD card, displayed a bitmap, used touch to draw on TFT. All worked as expected.

The second program used the GFX  and displayed graphics using drawing commands as well as text to the TFT display. It also compiled and worked without a problem.

hathach, 2020-09-24:
thank you for confirmation :)

PR #75 (2020-09-21): Fix build with esp32
fix #74