adafruit/Adafruit_SPIFlash #77: Adafruit_SPIFlashBase.h:35:28: warning: overflow in constant expression [-fpermissive]

MartinBzm, 2020-10-23:
Arduino: 1.8.13 (Windows 7), Board: "Arduino Mega 2560, ATmega2560 (Mega 2560)"
Modules used:
- DeekRobot 8122 RTC & SDcard module, connected with I2C
- TFT LCD 128x160 RGB with ST7735 chip, connected on ICSP header

When -not- using SPI but the SDcard on the  this error pops up:

F:\Projecten\Arduino\libraries\Adafruit_SPIFlash\src/Adafruit_SPIFlashBase.h:35:26: warning: integer overflow in expression [-Woverflow]
   SFLASH_BLOCK_SIZE = 64 * 1024,
                       ~~~^~~~~~
F:\Projecten\Arduino\libraries\Adafruit_SPIFlash\src/Adafruit_SPIFlashBase.h:35:28: warning: overflow in constant expression [-fpermissive]
   SFLASH_BLOCK_SIZE = 64 * 1024,
                            ^~~~
F:\Projecten\Arduino\libraries\Adafruit_SPIFlash\src/Adafruit_SPIFlashBase.h:35:28: warning: overflow in constant expression [-fpermissive]

This also happens in the example BreakoutST7735-160x128.

hathach, 2020-10-24:
Do you have the sketch that causes and step to reproduce the issue

hathach, 2020-11-12:
update: I see the enum is treated as int16_t therefore 64KB is overflow, will fix it in the next pr.

PR #79 (2020-11-12): fix warning due to default int size of 16 on avr mcu
fix #77
closed 2020-11-12 by hathach