adafruit/Adafruit_SPIFlash #82: Fails when using with Feather M0 Express starting with v3.3.0

brysonian, 2020-11-26:
**Describe the bug**
When using versions >= 3.3.0 the library silently fails to write to files. 

**Set up (mandatory)**
 - **PC & IDE** : Arduino 1.8.13 on macOS 10.15.6
 - **Board** : Feather M0 Express
 - **BSP**   : 1.6.4
 - **Sketch**: examples/SdFat_full_usage

**To Reproduce**
Steps to reproduce the behavior:
1. Install a version of the library >= 3.3.0 <= 3.3.3 (latest)
2. Connect Feather M0 Express
3. Open `SdFat_full_usage` example
4. Upload to board
5. See errors in output

**Serial Log**
Using 3.3.3
```
Adafruit SPI Flash FatFs Full Usage Example
Flash chip JEDEC ID: 0xC84015
Mounted filesystem!
Test directory not found, creating...
Error, failed to create directory!
```

Using 3.3.0
```
Adafruit SPI Flash FatFs Full Usage Example
Flash chip JEDEC ID: 0xC84015
Mounted filesystem!
Test directory not found, creating...
Created directory!
Creating deep folder structure...
Creating /test/foo/bar
Created directory!
Creating /test/foo/baz
Error, failed to create directory!
```

For reference, when using 3.2.2
```
Adafruit SPI Flash FatFs Full Usage Example
Flash chip JEDEC ID: 0xC84015
Mounted filesystem!
Test directory not found, creating...
Created directory!
Creating deep folder structure...
Creating /test/foo/bar
Created directory!
Creating /test/foo/baz
Created directory!
Opened file /test/test.txt for writing/appending...
Wrote to file /test/test.txt!
First line of test.txt: Hello world!

Total size of test.txt (bytes): 57
Current position in test.txt: 14
Available data to read in test.txt: 43
File name: use getName()
Is file a directory? No
Entire contents of test.txt:
Hello world!
Hello number: 123
Hello hex number: 0x7B
Listing children of directory /test:
- foo (directory)
- test.txt
Deleting /test/foo/test2.txt...
Deleted file!
Deleting /test directory and everything inside it...
Test directory was deleted!
Finished!
```

hathach, 2020-11-27:
hmm, I just pull out my feather M0 and test, mine running just well. Version 3.3.0 bump up the  speed of SPI on M0 to 24Mhz which is technically not supported by the MCU, but proves to work rather well on my local testing. But apparently it does not happen to all variant.

Would you mind to comment out this line, this will impose a limit of 12Mhz read speed for M0 https://github.com/adafruit/Adafruit_SPIFlash/blob/master/src/Adafruit_SPIFlashBase.cpp#L155

result would be
```
  //if (_flash_dev->is_fram) {
    rd_clock_speed = min(12000000, rd_clock_speed);
  //}
```

then test again to see if that work well

brysonian, 2020-11-30:
Yep that was the culprit, after this it works as expected.

ladyada, 2020-11-30:
@hathach please force 12mhz :)

PR #83 (2020-11-30): limit M0 read speed to 12 Mhz for reliability
fix #82
closed 2020-11-30 by hathach