adafruit/Adafruit_SPIFlash #84: Unable to use both SD.h and Adafruit_SPIFlash.h in the same sketch

DrItanium, 2020-12-03:
**Describe the bug**

I am unable to use the builtin SDCard slot and onboard SPI Flash on my Grand Central M4. I get compilation errors.
Including SD.h and Adafruit_SPIFlash.h yields different compilation errors depending on order of includes (SD.h before or after Adafruit_SPIFlash.h).

[SD_h_is_first.txt](https://github.com/adafruit/Adafruit_SPIFlash/files/5633669/SD_h_is_first.txt)
I have attached the full console output from each run, I don't know how useful they are since they are both quite massive. I made the reduced test cases directly in the sketch where I found the error.  

**Set up (mandatory)**
 - **PC & IDE** : Arduino 1.8.12 on Gentoo Linux amd64 
 - **Board** : Grand Central M4 
 - **BSP**   : Adafruit SAMD Boards v 1.6.4 with boards.txt modified to compile in gnu++17 mode
 - **Sketch**: see Custom Sketches section

**To Reproduce**
Steps to reproduce the behavior:
Run verify on each custom sketch below to see each error

**Compiler Error Output**
When SD.h comes after Adafruit_SPIFlash.h:

```
In file included from /home/jwscoggins/dev/sketches/i960_Prototype_Simulator/i960_Prototype_Simulator.ino:140:
/home/jwscoggins/dev/sketches/libraries/SD/src/SD.h:23: warning: "FILE_READ" redefined
   23 | #define FILE_READ O_READ
      | 
In file included from /home/jwscoggins/dev/sketches/libraries/SdFat_-_Adafruit_Fork/src/FatLib/FatLib.h:27,
                 from /home/jwscoggins/dev/sketches/libraries/SdFat_-_Adafruit_Fork/src/SdFat.h:33,
                 from /home/jwscoggins/dev/sketches/i960_Prototype_Simulator/i960_Prototype_Simulator.ino:138:
/home/jwscoggins/dev/sketches/libraries/SdFat_-_Adafruit_Fork/src/FatLib/ArduinoFiles.h:37: note: this is the location of the previous definition
   37 | #define FILE_READ O_RDONLY
      | 
In file included from /home/jwscoggins/dev/sketches/i960_Prototype_Simulator/i960_Prototype_Simulator.ino:140:
/home/jwscoggins/dev/sketches/libraries/SD/src/SD.h:24: warning: "FILE_WRITE" redefined
   24 | #define FILE_WRITE (O_READ | O_WRITE | O_CREAT | O_APPEND)
      | 
In file included from /home/jwscoggins/dev/sketches/libraries/SdFat_-_Adafruit_Fork/src/FatLib/FatLib.h:27,
                 from /home/jwscoggins/dev/sketches/libraries/SdFat_-_Adafruit_Fork/src/SdFat.h:33,
                 from /home/jwscoggins/dev/sketches/i960_Prototype_Simulator/i960_Prototype_Simulator.ino:138:
/home/jwscoggins/dev/sketches/libraries/SdFat_-_Adafruit_Fork/src/FatLib/ArduinoFiles.h:39: note: this is the location of the previous definition
   39 | #define FILE_WRITE (O_RDWR | O_CREAT | O_AT_END)
      | 
In file included from /home/jwscoggins/dev/sketches/i960_Prototype_Simulator/i960_Prototype_Simulator.ino:140:
/home/jwscoggins/dev/sketches/libraries/SD/src/SD.h:63:7: error: 'SdVolume' does not name a type; did you mean 'FatVolume'?
   63 |       SdVolume volume;
      |       ^~~~~~~~
      |       FatVolume
/home/jwscoggins/dev/sketches/libraries/SD/src/SD.h:71:37: error: 'SD_CHIP_SELECT_PIN' was not declared in this scope
   71 |       boolean begin(uint8_t csPin = SD_CHIP_SELECT_PIN);
      |                                     ^~~~~~~~~~~~~~~~~~
```
And when SD.h comes before Adafruit_SPIFlash.h

```
In file included from /home/jwscoggins/dev/sketches/i960_Prototype_Simulator/i960_Prototype_Simulator.ino:139:
/home/jwscoggins/dev/sketches/libraries/Adafruit_SPIFlash/src/Adafruit_SPIFlash.h:50:49: error: expected class-name before ',' token
   50 | class Adafruit_SPIFlash : public BaseBlockDriver, public Adafruit_SPIFlashBase {
      |                                                 ^

```

**Custom Sketches**
```
// SD.h after Adafruit_SPIFlash.h
#include <SdFat.h>
#include <Adafruit_SPIFlash.h>
#include <SD.h>
Adafruit_FlashTransport_QSPI flashTransport;
Adafruit_SPIFlash flash(&flashTransport);
void setup() {
    flash.begin();
}
void loop() {
    delay(100);
}
```

and 
```
// SD.h before Adafruit_SPIFlash.h
#include <SD.h>
#include <SdFat.h>
#include <Adafruit_SPIFlash.h>

Adafruit_FlashTransport_QSPI flashTransport;
Adafruit_SPIFlash flash(&flashTransport);
void setup() {
    flash.begin();
}
void loop() {
    delay(100);
}
```

ladyada, 2020-12-03:
dont use SD.h just use SdFat

DrItanium, 2020-12-03:
Good to know! Do any of the shipped examples show how to interact with both the SD card and QSPI flash at the same time? I didn't see any...

ladyada, 2020-12-03:
nope, only arcada - if you write an example please submit it as a PR :)

DrItanium, 2020-12-03:
Okey dokey. I'll take a look at the arcada library and go from there. Thanks :)

closed 2020-12-03 by DrItanium

DrItanium, 2020-12-04:
As a heads up, the arcada library helped me solve my problem. I was also able to make a succinct example from it for this library. Thanks again :D.