adafruit/Adafruit_SPIFlash #85: Adafrut_SPIFlash Lib with ESP32-S2 Arduino Core

nikthefix, 2020-12-03:
I've been trying to get some of the  examples to compile for the Saola board using external SPI Fram - a configuration which compiles and works well with SAMD21/51 boards as long as DMA spi reads are disabled. (https://github.com/adafruit/Adafruit_SPIFlash/issues/65)

For the Saola I uncomment the custom spi pin defines but on compile I get:

error: 'esp_partition_t' does not name a type; did you mean 'esp_ptr_internal'?
   esp_partition_t const *_partition;
   ^~~~~~~~~~~~~~~
   esp_ptr_internal

I get get the same error if I leave the default defines to target the internal flash partition with FATFS default partition scheme. 
Any pointers? (no pun).

hathach, 2020-12-04:
Hi Nick, please open issue using bug template and fill out all the details there, it helps us a lot in understanding and reproducing the issue.

hathach, 2021-02-01:
closed due to lack of response from user

closed 2021-02-01 by hathach