adafruit/Adafruit_SPIFlash #91: SdFat examples do not compile: Cannot convert 'Adafruit_SPIFlash*' to 'BlockDriver*' {aka 'SdSpiCard*'}

BriscoeTech, 2021-03-14:
Hello Everyone,

I am trying to setup my QT PY with a external flash chip soldered on the bottom side, and I am running into some compiling problem for all the SdFat Examples in this library. My flash chip is working, and am able to run flash_info and flash_eraser without any issues.

I keep getting this compile error whenever fatfs.begin(&flash) is called in the SdFat example programs

> cannot convert 'Adafruit_SPIFlash*' to 'BlockDriver*' {aka 'SdSpiCard*'}

Current Libraries and Boards being used
Adafruit SAMD Boards v1.6.5
Adafruit_SdFat at version 1.2.3
Adafruit_SPIFlash at version 3.4.1

It is as if a typecast is needed, so I tried typecasting to fatfs.begin((BlockDriver*)&flash) which compiles, but causes the code to hang and is not the solution.

Has something fundamentally been changed in the library that needs the examples updated?

Thanks for your help!

hathach, 2021-03-15:
please provide all the information in the bug template, which sketch you are compiling, best to be one of the stock example of this repo. Essentially your detail set-up and how to reproduce the issue.

BriscoeTech, 2021-03-15:
My origional post does cover all the details needed to reproduce the problem. _All of the stock SdFat examples_ in this repository are having this problem as they all call fatfs.begin(&flash), but first discovered the problem trying to use SdFat_format.ino

Setup is with all the library versions listed above with a QT PY board.

hathach, 2021-03-16:
next time, please fill in the bug template, those are mandatory when opening an issue. Back to the issue, I just tried out an example and it compiles just fine. The ci rebuild also compile without issue 
https://github.com/adafruit/Adafruit_SPIFlash/runs/2115085814?check_suite_focus=true

Arduino library probably pick the original SdFat instead of the adafruit's fork (which include the modification to address that error). Can you post your compile log where arduino output which libraries is used

BriscoeTech, 2021-03-16:
Hmmmm that's interesting, I went through and double checked I do not have griemans sd fat installed anymore and only have the Adafruit port v1.2.3. I deleted griemans yesterday thinking that could have been a problem.

Here is the full compile log: [compileLog_SpiFlash.txt](https://github.com/adafruit/Adafruit_SPIFlash/files/6146331/compileLog_SpiFlash.txt)

Looking though the output it does seem like it is using Adafruit_SdFat [line 74]

hathach, 2021-03-16:
I couldn't reproduce the issue, can you remove the current Adafruit Sdfat, then git clone or installed it again

BriscoeTech, 2021-03-18:
There we go, that seemed to do it! Everything is compiling now and running :-)

I deleted the existing library folder and then did a git clone, and it started worked right away. Confused, I diffed the cloned folder with what I deleted, and there was no difference. When I did the clone the default folder name was SdFat, I changed it to Adafruit_SdFat to match what I originally deleted and it continued to work.

I am thinking that the Arduino IDE cached something wrong, like possibly Griemens compiled version. I think by deleting all the different SdFat libraries and then cloning the folder with a different path finally got arduino to recompile things properly.

Thank you for your help! Got the msc working like a charm now.

closed 2021-03-18 by BriscoeTech

hathach, 2021-03-18:
I am glad that works for you.