adafruit/Adafruit_SPIFlash #94: flash device definition appears to be wrong

CreativeRobotics, 2021-06-09:
I tried using the W25Q80DL in a project and the JDEC ID didn't match the one in flash_devices.h It reports the memory_type as 0x40 not 0x60.

I was using a SAMD21 with the W25Q80DL flash chip from Winbond. I was testing using various spi Flash examples from this library.

I also noticed (unless I missed something) that the max speed is wrong. The datasheed list two device versions, one with a speed of 104MHz and the other with 80MHz. The DL variant has a top speed of 80MHz

W25Q80DV 104MHz
W25Q80DL 80MHz

To fix the problem I changed this definition in flash_devices.h:

```
#define W25Q80DL                                                               \
  {                                                                            \
    .total_size = (1UL << 20), /* 1 MiB */                                     \
        .start_up_time_us = 5000, .manufacturer_id = 0xef,                     \
    .memory_type = 0x60, .capacity = 0x14, .max_clock_speed_mhz = 104,         \
    .quad_enable_bit_mask = 0x02, .has_sector_protection = false,              \
    .supports_fast_read = true, .supports_qspi = true,                         \
    .supports_qspi_writes = false, .write_status_register_split = false,       \
    .single_status_byte = false, .is_fram = false,                             \
  }
```
	
To this:

```
#define W25Q80DL                                                               \
  {                                                                            \
    .total_size = (1UL << 20), /* 1 MiB */                                     \
        .start_up_time_us = 5000, .manufacturer_id = 0xef,                     \
    .memory_type = 0x40, .capacity = 0x14, .max_clock_speed_mhz = 80,         \
    .quad_enable_bit_mask = 0x02, .has_sector_protection = false,              \
    .supports_fast_read = true, .supports_qspi = true,                         \
    .supports_qspi_writes = false, .write_status_register_split = false,       \
    .single_status_byte = false, .is_fram = false,                             \
  }
```

hathach, 2021-06-10:
great spot, can you submit an PR for this ?

CreativeRobotics, 2021-06-10:
Done. I added a couple of other defice definitions, and a new example as well if they are of use.

hathach, 2021-06-10:
> Done. I added a couple of other defice definitions, and a new example as well if they are of use.

Thanks, feel free to add any devices that you could tested. Since it does not occupies any spaces at all. I hope it will grow large enough for most users.

hathach, 2021-06-11:
closed by #95

closed 2021-06-11 by hathach