adafruit/Adafruit_SPIFlash #104: nRF52 QSPI transport fails to operate in 1BIT I/O mode

lyusupov, 2021-09-08:
### Operating System

Linux

### IDE version

Arduino 1.6+

### Board

nRF52840

### BSP version

1.0.0

### SPIFlash Library version

Latest

### Sketch

[msc_external_flash.ino](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/blob/master/examples/MassStorage/msc_external_flash/msc_external_flash.ino) from Adafruit_TinyUSB_Arduino examples.

### What happened ?

_Adafruit_FlashTransport_QSPI_NRF.cpp_ always uses 'hard-coded' 4-bits data R/W transfer:

```
      .prot_if = {.readoc = NRF_QSPI_READOC_READ4O, // 0x6B read command
                  .writeoc = NRF_QSPI_WRITEOC_PP4O, // 0x32 write command
```

regardless of the 

```
.supports_qspi = false,
```

defined in the _flash_devices.h_

The _supports_qspi _ variable has an effect on STATUS REGISTER value, but not on the R/W COMMANDs data width.

To maintain compatibility with 1BIT data transfer mode I would suggest to initialize the nRF52 QSPI transport as follows:

```
      .prot_if = {.readoc = NRF_QSPI_READOC_FASTREAD, // 0x0B read command
                  .writeoc = NRF_QSPI_WRITEOC_PP, // 0x02 write command
```

then update the data transfer method if the  .supports_qspi is true for flash IC been detected.

### How to reproduce ?

build and run sketch listed above.

hathach, 2021-09-09:
Can you submit an PR for this, to be honest, this 1bit IO is not very useful. People should just use spi instead

lyusupov, 2021-09-09:
I am not enough famliar with the library API to submit an **elegant** solution. That's why I rely on your discretion.

When every SPIM module of nRF52 is in use - there is no other way to operate SPI flash other than being connected to QSPI transport.
nRF52 QSPI module does work independent of SPIM modules.

hathach, 2021-09-09:
since Adafruit does not make any board that need this 1bit IO communication, there is no motivation for me to write such as code. If you are worried about the code, just make an PR as best as you could. I could help shaping it with review.