adafruit/Adafruit_SPIFlash #110: Error, f_mount failed with error code: 13 using Teensy 4.0 and S25FL064L

abhignay, 2022-01-31:
### Operating System

Linux

### IDE version

Arduino 1.8.19

### Board

Teensy 4.0, Teensy 3.2

### BSP version

1.0.0

### SPIFlash Library version

3.8

### Sketch

examples/SdFat_Format

### What happened ?

I am using an S25FL064L and Teensy 4.0. When I run the SdFat_Format file example it gives me the following error:
```
Adafruit SPI Flash FatFs Format Example
Flash chip JEDEC ID: 0x16017
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
This sketch will ERASE ALL DATA on the flash chip and format it with a new filesystem!
Type OK (all caps) and press enter to continue.
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
Creating and formatting FAT filesystem (this takes ~60 seconds)...
Error, f_mount failed with error code: 13
```

When I run the same file with the same flash chip on the Teensy 3.2 it works it doesn't give me error code 13

Nothing changed between these two tests except for the microcontroller being used. The example/flash_info gives the correct output for both flash chips using both microcontrollers. So for me atleast it looks like the Teensy 4.0 is not working with the Adafruit Flash library

### How to reproduce ?

1. Wire up your Teensy 4.0 and S25Fl064L using the SPI protocol
2. Run the SdFat_Format.ino file

### Screenshots

Error 13 with Teensy 4.0
[image]

Works with Teensy 3.2
[image]

hathach, 2022-02-01:
Unfortunately I am not familiar enough with teensy 4.0 as well as having the S25Fl064L to help you put. Hope that other with similar set up could help

wh201906, 2022-03-16:
Hello, I got the same error with the same SdFat_Format example on Seeeduino Xiao(SAMD21G18, ARM Cortex-M0)

hathach, 2022-05-05:
closed since hardware is not available to us therefore no way to add support for this.

closed 2022-05-05 by hathach