adafruit/Adafruit_SPIFlash #115: run example Error in esp32-c3

QGB, 2022-04-25:
### Operating System

Windows 10

### IDE version

1.8.19

### Board

esp32-c3

### BSP version

==

### SPIFlash Library version

3.9.1

### Sketch

SdFat_full_usage.ino

[image]

### What happened ?

compiling SdFat_full_usage.ino success.

But can not boot example in board

### How to reproduce ?

```
项目使用了 203746 字节，占用了 (15%) 程序存储空间。最大为 1310720 字节。
全局变量使用了10200字节，(3%)的动态内存，余留317480字节局部变量。最大为327680字节。
esptool.py v3.1
Serial port COM16
Connecting....
Chip is unknown ESP32-C3 (revision 3)
Features: Wi-Fi
Crystal is 40MHz
MAC: 60:55:f9:72:d7:48
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00000000 to 0x00002fff...
Flash will be erased from 0x00010000 to 0x00042fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 683.0 kbit/s)...
Hash of data verified.
Flash params set to 0x022f
Compressed 11776 bytes to 8489...
Writing at 0x00000000... (100 %)
Wrote 11776 bytes (8489 compressed) at 0x00000000 in 0.9 seconds (effective 104.5 kbit/s)...
Hash of data verified.
Compressed 206080 bytes to 114966...
Writing at 0x00010000... (12 %)
Writing at 0x0001ab26... (25 %)
Writing at 0x0002075a... (37 %)
Writing at 0x000271d7... (50 %)
Writing at 0x0002d2ca... (62 %)
Writing at 0x00033371... (75 %)
Writing at 0x0003b7e7... (87 %)
Writing at 0x00042231... (100 %)
[... 144 lines cut ...]
3fc92c10: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x3fc8d000 0x42003000 0x42003e3c
3fc92c30: 0x00000000 0x00000000 0x00000000 0x403869d0 0x00000000 0x00000000 0x00000000 0x00000000
3fc92c50: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x27146414 0x01010107 0x00000001 0x00000000
3fc92c70: 0xffffffff 0x7fefffff 0x00000000 0x3fc00000 0x00000000 0x40300000 0x00000000 0x3fe00000
3fc92c90: 0x00000000 0x3ff80000 0x636f4361 0x3fd287a7 0x8b60c8b3 0x3fc68a28 0x509f79fb 0x3fd34413
3fc92cb0: 0x00000000 0x3ff00000 0x00000000 0x40240000 0x00000000 0x401c0000 0x00000000 0x40140000
3fc92cd0: 0x00000000 0x43500000 0x00000010 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0x00000154 0x3fc92970
3fc92cf0: 0x3fc8cbf4 0x3fc8c4dc 0x3fc8c4dc 0x3fc92cec 0x3fc8c4d4 0x00000018 0x902e4c78 0x449956b8
3fc92d10: 0x3fc92cec 0x00000000 0x00000001 0x3fc90ce8 0x706f6f6c 0x6b736154 0xac504800 0x00fd5270
3fc92d30: 0x00000000 0x3fc92ce0 0x00000001 0x00000000 0x00000000 0x00000000 0x00000000 0x3fc8d58c
3fc92d50: 0x3fc8d5f4 0x3fc8d65c 0x00000000 0x00000000 0x00000001 0x00000000 0x00000000 0x00000000
3fc92d70: 0x420118f6 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc92d90: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc92db0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc92dd0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc92df0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc92e10: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc92e30: 0x00000000 0x00000000 0x00000000 0x3fc90000 0x00000054 0x00000000 0x3fc92e44 0x00000000
3fc92e50: 0x00000000 0x00000000 0x3fc92e5c 0xffffffff 0x3fc92e5c 0x3fc92e5c 0x00000000 0x3fc92e70
3fc92e70: 0xffffffff 0x3fc92e70 0x3fc92e70 0x00000001 0x00000001 0x00000000 0x5300ffff 0x00000000

ELF file SHA256: 0000000000000000

Rebooting...
```

hathach, 2022-04-25:
C3 may not be supported yet, I will check it later. Please put all the log into txt file (attached) for readibility

hathach, 2022-05-05:
just tested, everything work fine with C3, you need to select the partition scheme to include fatfs for it to work since it is what this repo focus on.

[image]

closed 2022-05-05 by hathach