adafruit/Adafruit_SPIFlash #116: undefined reference to `Adafruit_FlashTransport_QSPI::Adafruit_FlashTransport_QSPI()'

cantonfoshan, 2022-05-04:
### Operating System

MacOS

### IDE version

1.8.13

### Board

AdaFruit e-ink sd card

### BSP version

2.13

### SPIFlash Library version

1.5.1

### Sketch as attached file if not stock example

/***************************************************
  Adafruit invests time and resources providing this open source code,
  please support Adafruit and open-source hardware by purchasing
  products from Adafruit!

  Written by Limor Fried/Ladyada for Adafruit Industries.
  MIT license, all text above must be included in any redistribution
 ****************************************************/

#include <SPI.h>
#include "Adafruit_EPD.h"

#include <SdFat.h>
#include <Adafruit_FlashTransport.h>
#include <Adafruit_FlashTransport_QSPI.h>
#include <Adafruit_SPIFlashBase.h>
#include <Adafruit_FlashCache.h>
#include <flash_devices.h>
#include <Adafruit_SPIFlash.h>

#define EPD_CS     10
#define EPD_DC      9
#define SRAM_CS     11
#define EPD_RESET   5 // can set to -1 and share with microcontroller Reset!
#define EPD_BUSY    7 // can set to -1 to not use a pin (will wait a fixed delay)

// Uncomment the following line if you are using 1.54" EPD with IL0373
//Adafruit_IL0373 display(152, 152, EPD_DC, EPD_RESET, EPD_CS, SRAM_CS, EPD_BUSY);
// Uncomment the following line if you are using 1.54" EPD with SSD1680
//Adafruit_SSD1680 display(152, 152, EPD_DC, EPD_RESET, EPD_CS, SRAM_CS, EPD_BUSY);
// Uncomment the following line if you are using 1.54" EPD with SSD1608
//Adafruit_SSD1608 display(200, 200, EPD_DC, EPD_RESET, EPD_CS, SRAM_CS, EPD_BUSY);
// Uncomment the following line if you are using 1.54" EPD with SSD1681
//Adafruit_SSD1681 display(200, 200, EPD_DC, EPD_RESET, EPD_CS, SRAM_CS, EPD_BUSY);
// Uncomment the following line if you are using 1.54" EPD with UC8151D
//Adafruit_UC8151D display(152, 152, EPD_DC, EPD_RESET, EPD_CS, SRAM_CS, EPD_BUSY);

//Uncomment the following line if you are using 2.13" EPD with SSD1680
Adafruit_SSD1680 display(250, 122, EPD_DC, EPD_RESET, EPD_CS, SRAM_CS, EPD_BUSY);

// Uncomment the following line if you are using 2.13" EPD with SSD1675
//Adafruit_SSD1675 display(250, 122, EPD_DC, EPD_RESET, EPD_CS, SRAM_CS, EPD_BUSY);

// Uncomment the following line if you are using 2.13" EPD with SSD1675B
//Adafruit_SSD1675B display(250, 122, EPD_DC, EPD_RESET, EPD_CS, SRAM_CS, EPD_BUSY);

// Uncomment the following line if you are using 2.13" EPD with UC8151D
//Adafruit_UC8151D display(212, 104, EPD_DC, EPD_RESET, EPD_CS, SRAM_CS, EPD_BUSY);
[... 241 lines cut ...]
    if (entry.isDirectory()) {
      Serial.println("/");
      printDirectory(entry, numTabs + 1);
    } else {
      // files have sizes, directories do not
      Serial.print("\t\t");
      Serial.println(entry.size(), DEC);
    }
    entry.close();
  }
}

### What happened ?

Adafruit_FlashTransport_QSPI flashTransport;
Adafruit_SPIFlash flash(&flashTransport);
FatFileSystem fatfs;

### How to reproduce ?

Try to compile this code

Adafruit_FlashTransport_QSPI flashTransport;
Adafruit_SPIFlash flash(&flashTransport);
FatFileSystem fatfs;

ladyada, 2022-05-04:
what is the microcontroller board

hathach, 2022-05-05:
also please provide 
1. the exact board you are using and
2. attached arduino compile log as txt file

ricardo192, 2022-08-01:
I have this same issue. with the itsy bitsy m4

hathach, 2022-08-01:
can you try to compile to see if it compile with Arduino IDE instead of PIO ?

ricardo192, 2022-08-01:
Sure, but it's the same compiler and same code so it shouldn't make a difference.

ricardo192, 2022-08-01:
weird, it compiles

ladyada, 2022-08-01:
we agree... but the CI test works great here https://github.com/adafruit/Adafruit_SPIFlash/actions
so there is evidence that it works in arduino and not in PIO
cc @ivankravets

hathach, 2022-08-02:
@ricardo192 PIO parser may or may not pick up the same CFlags as Arduino, which does affect the compiling process. So you need to check if PIO parser does it job, and/or need to manual add config if needed.

valeros, 2022-08-02:
Hi @ricardo192! It's strange, but I cannot reproduce the issue. Could you please provide a minimal but complete PlatformIO project to reproduce the issue?

ricardo192, 2022-08-02:
[teststuf copy.zip](https://github.com/adafruit/Adafruit_SPIFlash/files/9245791/teststuf.copy.zip)

valeros, 2022-08-03:
@ricardo192 In a nutshell, PlatformIO supports many types of libraries, including Arduino ones. For some reason you decided to manually add the `Adafruit SPIFlash` library to the `lib` folder of you project instead of specifying it as dependency in your `platformio.ini`. At the same time you also changed the structure of the library (moved the source files from the `src` folder to the root of the library). This change you made forced PlatformIO to handle the `flash` library in your project as a generic library and hence omit special compilation logic required for the Arduino libraries.

To fix the problem you can either specify the library as a dependency, for example as a Git repository with a specific tag:
```
[env:adafruit_itsybitsy_m4]
platform = atmelsam
framework = arduino
board = adafruit_itsybitsy_m4
lib_deps = 
    https://github.com/adafruit/Adafruit_SPIFlash#3.11.0
```

or just move the library sources to the `src` folder, e.g. `lib/flash/src/Adafruit_SPIFlash.cpp`

ricardo192, 2022-08-03:
@valeros My bad! I have been doing this for a long while now (for at least like 20 Arduino libraries) and not have had a single issue with it until just now. also. I only just now got it to compile when I used the library manager. and doing it in the lib folder 'properly' still doesn't work for some reason. thanks for the help!

hathach, 2023-12-18:
closed, this is caused by pio config rather than the library itself

closed 2023-12-18 by hathach

skyswordx, 2024-07-31:
when I try to run the example flash_info， the same error happens while linking
```shell
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32H750VB.html
PLATFORM: ST STM32 (17.4.0) > STM32H750VBT6 (1024k RAM. 128k Flash)
HARDWARE: STM32H750VBT6 480MHz, 1MB RAM, 128KB Flash
DEBUG: Current (stlink) External (blackmagic, jlink, stlink)
PACKAGES:
 - framework-arduinoststm32 @ 4.20701.0 (2.7.1)
 - framework-cmsis @ 2.50900.0 (5.9.0)
 - toolchain-gccarmnoneeabi @ 1.120301.0 (12.3.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 15 compatible libraries
Scanning dependencies...
Dependency Graph
|-- Adafruit SPIFlash @ 4.3.4+sha.160b5ff
|-- SdFat - Adafruit Fork @ 2.2.3
|-- SPI @ 1.1.0
Building in release mode
Compiling .pio\build\genericSTM32H750VB\src\main.cpp.o
Compiling .pio\build\genericSTM32H750VB\lib7c6\SdFat - Adafruit Fork\FatLib\FatName.cpp.o
Compiling .pio\build\genericSTM32H750VB\lib7c6\SdFat - Adafruit Fork\FatLib\FatPartition.cpp.o
Compiling .pio\build\genericSTM32H750VB\lib7c6\SdFat - Adafruit Fork\FatLib\FatVolume.cpp.o
Compiling .pio\build\genericSTM32H750VB\lib7c6\SdFat - Adafruit Fork\FreeStack.cpp.o
Compiling .pio\build\genericSTM32H750VB\lib7c6\SdFat - Adafruit Fork\FsLib\FsFile.cpp.o
Compiling .pio\build\genericSTM32H750VB\lib7c6\SdFat - Adafruit Fork\FsLib\FsNew.cpp.o
Compiling .pio\build\genericSTM32H750VB\lib7c6\SdFat - Adafruit Fork\FsLib\FsVolume.cpp.o
Compiling .pio\build\genericSTM32H750VB\lib7c6\SdFat - Adafruit Fork\MinimumSerial.cpp.o
Compiling .pio\build\genericSTM32H750VB\lib7c6\SdFat - Adafruit Fork\SdCard\SdCardInfo.cpp.o
Compiling .pio\build\genericSTM32H750VB\lib7c6\SdFat - Adafruit Fork\SdCard\SdSpiCard.cpp.o
Compiling .pio\build\genericSTM32H750VB\lib7c6\SdFat - Adafruit Fork\SdCard\SdioTeensy.cpp.o
Compiling .pio\build\genericSTM32H750VB\lib7c6\SdFat - Adafruit Fork\SpiDriver\SdSpiArtemis.cpp.o
Compiling .pio\build\genericSTM32H750VB\lib7c6\SdFat - Adafruit Fork\SpiDriver\SdSpiChipSelect.cpp.o
Compiling .pio\build\genericSTM32H750VB\lib7c6\SdFat - Adafruit Fork\SpiDriver\SdSpiDue.cpp.o
Compiling .pio\build\genericSTM32H750VB\lib7c6\SdFat - Adafruit Fork\SpiDriver\SdSpiESP.cpp.o
Compiling .pio\build\genericSTM32H750VB\lib7c6\SdFat - Adafruit Fork\SpiDriver\SdSpiParticle.cpp.o
Compiling .pio\build\genericSTM32H750VB\lib7c6\SdFat - Adafruit Fork\SpiDriver\SdSpiSTM32.cpp.o
Compiling .pio\build\genericSTM32H750VB\lib7c6\SdFat - Adafruit Fork\SpiDriver\SdSpiSTM32Core.cpp.o
Compiling .pio\build\genericSTM32H750VB\lib7c6\SdFat - Adafruit Fork\SpiDriver\SdSpiTeensy3.cpp.o
Compiling .pio\build\genericSTM32H750VB\lib7c6\SdFat - Adafruit Fork\common\FmtNumber.cpp.o
Compiling .pio\build\genericSTM32H750VB\lib7c6\SdFat - Adafruit Fork\common\FsCache.cpp.o
Compiling .pio\build\genericSTM32H750VB\lib7c6\SdFat - Adafruit Fork\common\FsDateTime.cpp.o
Compiling .pio\build\genericSTM32H750VB\lib7c6\SdFat - Adafruit Fork\common\FsName.cpp.o
Compiling .pio\build\genericSTM32H750VB\lib7c6\SdFat - Adafruit Fork\common\FsStructs.cpp.o
Compiling .pio\build\genericSTM32H750VB\lib7c6\SdFat - Adafruit Fork\common\FsUtf.cpp.o
Compiling .pio\build\genericSTM32H750VB\lib7c6\SdFat - Adafruit Fork\common\PrintBasic.cpp.o
Compiling .pio\build\genericSTM32H750VB\lib7c6\SdFat - Adafruit Fork\common\upcase.cpp.o
Compiling .pio\build\genericSTM32H750VB\lib7c6\SdFat - Adafruit Fork\iostream\StdioStream.cpp.o
Compiling .pio\build\genericSTM32H750VB\lib7c6\SdFat - Adafruit Fork\iostream\StreamBaseClass.cpp.o
Compiling .pio\build\genericSTM32H750VB\lib7c6\SdFat - Adafruit Fork\iostream\istream.cpp.o
Compiling .pio\build\genericSTM32H750VB\lib7c6\SdFat - Adafruit Fork\iostream\ostream.cpp.o
Compiling .pio\build\genericSTM32H750VB\lib40a\Adafruit SPIFlash\Adafruit_FlashCache.cpp.o
Compiling .pio\build\genericSTM32H750VB\lib40a\Adafruit SPIFlash\Adafruit_SPIFlash.cpp.o
Compiling .pio\build\genericSTM32H750VB\lib40a\Adafruit SPIFlash\Adafruit_SPIFlashBase.cpp.o
Compiling .pio\build\genericSTM32H750VB\lib40a\Adafruit SPIFlash\esp32\Adafruit_FlashTransport_ESP32.cpp.o
Compiling .pio\build\genericSTM32H750VB\lib40a\Adafruit SPIFlash\qspi\Adafruit_FlashTransport_QSPI_NRF.cpp.o
Compiling .pio\build\genericSTM32H750VB\lib40a\Adafruit SPIFlash\qspi\Adafruit_FlashTransport_QSPI_SAMD.cpp.o
Compiling .pio\build\genericSTM32H750VB\lib40a\Adafruit SPIFlash\rp2040\Adafruit_FlashTransport_RP2040.cpp.o
Compiling .pio\build\genericSTM32H750VB\lib40a\Adafruit SPIFlash\spi\Adafruit_FlashTransport_SPI.cpp.o
Compiling .pio\build\genericSTM32H750VB\FrameworkArduino\HardwareSerial.cpp.o
Compiling .pio\build\genericSTM32H750VB\FrameworkArduino\IPAddress.cpp.o
Compiling .pio\build\genericSTM32H750VB\FrameworkArduino\Print.cpp.o
Compiling .pio\build\genericSTM32H750VB\FrameworkArduino\RingBuffer.cpp.o
Compiling .pio\build\genericSTM32H750VB\FrameworkArduino\Stream.cpp.o
Compiling .pio\build\genericSTM32H750VB\FrameworkArduino\Tone.cpp.o
Compiling .pio\build\genericSTM32H750VB\FrameworkArduino\USBSerial.cpp.o
Compiling .pio\build\genericSTM32H750VB\FrameworkArduino\VirtIOSerial.cpp.o
Compiling .pio\build\genericSTM32H750VB\FrameworkArduino\WInterrupts.cpp.o
Compiling .pio\build\genericSTM32H750VB\FrameworkArduino\WMath.cpp.o
[... 142 lines cut ...]
void setup() {
  // put your setup code here, to run once:
    Serial.begin(115200);

}

void loop() {
  // put your main code here, to run repeatedly:
  Serial.println("Hello World! stm32duino is working!");
}
```
here is my pio config file
```ini
[platformio]

[env:genericSTM32H750VB]
platform = ststm32
board = genericSTM32H750VB
framework = arduino
debug_tool = stlink
upload_protocol = stlink

lib_deps = 
	https://github.com/adafruit/Adafruit_SPIFlash.git
```

skyswordx, 2024-07-31:
> @ricardo192 PIO parser may or may not pick up the same CFlags as Arduino, which does affect the compiling process. So you need to check if PIO parser does it job, and/or need to manual add config if needed.
My apologies for the interruption. I am new to using PIO and was wondering what configuration I should add to make a manual call.