adafruit/Adafruit_SPIFlash #120: ESP32 use external SPI flash cause Core 1 panic'ed (LoadProhibited)

sistik, 2022-08-08:
### Operating System

Windows 10

### IDE version

PlatformIO 6.1.3

### Board

Adafruit ESP32 Feather

### BSP version

not defined

### SPIFlash Library version

3.11.0

### Sketch as attached file if not stock example

```
#include <Arduino.h>
#include <Adafruit_SPIFlash.h>

Adafruit_FlashTransport_SPI flashTransport(5, SPI);
Adafruit_SPIFlash flash(&flashTransport);

void setup() {
    flash.begin();
}

void loop() {
    sleep(1000);
}
```

### What happened ?

There is redefined method 'Adafruit_SPIFlashBase::begin' for all instances. This redefinition should be used only for **module internal** flash not for **SPI** flash.
Problematic part is: `_flash_dev = ((Adafruit_FlashTransport_ESP32 *)_trans)->getFlashDevice();` where _trans is Adafruit_FlashTransport_SPI.
This should not be use when I am using SPI flash. When I **temporary** removed `defined(ARDUINO_ARCH_ESP32) ||` from file Adafruit_SPIFlashBaes.cpp there was no more panic and library was able to read size of external SPI flash.

Crash LOG from console:
```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400ec263  PS      : 0x00060e30  A0      : 0x800d13e8  A1      : 0x3ffb27a0  
A2      : 0x00000003  A3      : 0x00000001  A4      : 0x00000000  A5      : 0x00000020  
A6      : 0x00000004  A7      : 0x3ffc1a78  A8      : 0x3ffc1108  A9      : 0x009c1001  
A10     : 0x3ffc1120  A11     : 0x00000021  A12     : 0x00000021  A13     : 0x000000ff  
A14     : 0x000000ff  A15     : 0x3ffc1634  SAR     : 0x0000000e  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000001b  LBEG    : 0x400d2109  LEND    : 0x400d2118  LCOUNT  : 0x00000000  

Backtrace:0x400ec260:0x3ffb27a00x400d13e5:0x3ffb27c0 0x400d1365:0x3ffb27e0 0x400d0f94:0x3ffb2800 0x400d26a6:0x3ffb2820
```

### How to reproduce ?

Start provided code.

hathach, 2022-08-09:
thank you for the issue, could you please post the link to the problematic part to make sure we don't look at the wrong line of code.

sistik, 2022-08-09:
Here is place where is problematic method called:
```c++
bool Adafruit_SPIFlash::begin(SPIFlash_Device_t const *flash_devs,
                              size_t count) {
  bool ret = Adafruit_SPIFlashBase::begin(flash_devs, count);

  // Use cache if not FRAM
  if (_flash_dev && !_flash_dev->is_fram) {
    // new cache object if not already
    if (!_cache) {
      _cache = new Adafruit_FlashCache();
    }
  }

  return ret;
}
```

Here is the implementation of that method:
```c++
#if defined(ARDUINO_ARCH_ESP32) || defined(ARDUINO_ARCH_RP2040)

// For ESP32 and RP2040 the SPI flash is already detected and configured
// We could skip the initial sequence
bool Adafruit_SPIFlashBase::begin(SPIFlash_Device_t const *flash_devs,
                                  size_t count) {
  (void)flash_devs;
  (void)count;

  if (_trans == NULL) {
    return false;
  }

  _trans->begin();

#if defined(ARDUINO_ARCH_ESP32)
  _flash_dev = ((Adafruit_FlashTransport_ESP32 *)_trans)->getFlashDevice();
#elif defined(ARDUINO_ARCH_RP2040)
  _flash_dev = ((Adafruit_FlashTransport_RP2040 *)_trans)->getFlashDevice();
#endif

  return true;
}

#else

/// List of all possible flash devices used by Adafruit boards
static const SPIFlash_Device_t possible_devices[] = {
    // Main devices used in current Adafruit products
    GD25Q16C, GD25Q32C, GD25Q64C, S25FL116K, S25FL216K,

    // Only a handful of production run
    W25Q16FW, W25Q64JV_IQ,

    // Fujitsu FRAM
    MB85RS64V, MB85RS1MT, MB85RS2MTA, MB85RS4MT,

    // Other common flash devices
    W25Q16JV_IQ, W25Q32JV_IQ, AT25SF041, AT25DF081A};

/// Flash device list count
enum {
  EXTERNAL_FLASH_DEVICE_COUNT =
      sizeof(possible_devices) / sizeof(possible_devices[0])
};

static SPIFlash_Device_t const *findDevice(SPIFlash_Device_t const *device_list,
                                           int count,
                                           uint8_t const (&jedec_ids)[4]) {
  for (uint8_t i = 0; i < count; i++) {
[... 122 lines cut ...]
  } else if (_flash_dev->total_size > 64UL * 1024) {
    addr_byte = 3;
  } else {
    addr_byte = 2;
  }

  _trans->setAddressLength(addr_byte);

  // Turn off sector protection if needed
  //  if (_flash_dev->has_sector_protection)
  //  {
  //    writeEnable();
  //
  //    uint8_t data[1] = {0x00};
  //    QSPI0.writeCommand(QSPI_CMD_WRITE_STATUS, data, 1);
  //  }

  writeDisable();
  waitUntilReady();

  return true;
}

#endif // ARDUINO_ARCH_ESP32
```

hathach, 2022-08-09:
hmm, are you adding additional SPI flash to the feather ESP32 ? Could you provide info on the SPI flash (which breakout, chip). Indeed This library is mainly for used with module flash . I didn't test with this scenario. If you already made changes, maybe make an PR for review.

hathach, 2022-08-10:
I see, thank for the issue, though currently I don't have time to test this out and make the changes. I will do it later.

Yveaux, 2023-09-27:
Hi @hathach any update on this issue?
I ran into the same issue for rp2040 which uses the same construct as esp32.
Basically the Adafruit_FlashTransport_SPI gets cast to either Adafruit_FlashTransport_ESP32 or Adafruit_FlashTransport_RP2040 to get the flash device, but these objects are not the same.
There should be no need for this cast; it can be handled by the different transport implementations.

Jay7Kay, 2023-11-02:
Any update on the issue? I apparently run in to the same problem and I don't have the skills to fix it myself so any form of help would be appreciated.

hathach, 2023-12-18:
closed since currently this library is implemented to only support on-module flash device on both esp32 and rp2040. It does not supported additional external flash device on those platforms

quizque, 2024-05-22:
If anyone else comes across this problem and needs a solution, setting `#if defined(ARDUINO_ARCH_ESP32) || defined(ARDUINO_ARCH_RP2040)` to `#if false` on this line https://github.com/adafruit/Adafruit_SPIFlash/blob/160b5ff65347a741954566a47d170e26c5c847c4/src/Adafruit_SPIFlashBase.cpp#L64 makes it work. Not sure why support isn't being added when all the tutorials for external flash memory on the Adafruit website link to this library.