adafruit/Adafruit_SPIFlash #129: Compatibility issue with Adafruit_ILI0341 and Adafruit_SPIFlash

anttike, 2022-10-18:
### Operating System

MacOS

### IDE version

1.8.13

### Board

nRF52840 express

### BSP version

not sure

### SPIFlash Library version

GitHub latest

### Sketch as attached file if not stock example

#include <Adafruit_ILI9341.h>
#include <Adafruit_SPIFlash.h>

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}

### What happened ?

Won't compile. Msgs:

/Users/antti/Library/Arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: libraries/Adafruit_BusIO/Adafruit_SPIDevice.cpp.o: in function `Adafruit_SPIDevice::Adafruit_SPIDevice(signed char, unsigned long, BitOrder, unsigned char, SPIClass*)':
/Users/antti/Documents/Arduino/libraries/Adafruit_BusIO/Adafruit_SPIDevice.cpp:17: multiple definition of `Adafruit_SPIDevice::Adafruit_SPIDevice(signed char, unsigned long, BitOrder, unsigned char, SPIClass*)'; libraries/Adafruit_BusIO/Adafruit_SPIDevice 2.cpp.o:/Users/antti/Documents/Arduino/libraries/Adafruit_BusIO/Adafruit_SPIDevice 2.cpp:17: first defined here
/Users/antti/Library/Arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: libraries/Adafruit_BusIO/Adafruit_SPIDevice.cpp.o: in function `Adafruit_SPIDevice::Adafruit_SPIDevice(signed char, unsigned long, BitOrder, unsigned char, SPIClass*)':
/Users/antti/Documents/Arduino/libraries/Adafruit_BusIO/Adafruit_SPIDevice.cpp:17: multiple definition of `Adafruit_SPIDevice::Adafruit_SPIDevice(signed char, unsigned long, BitOrder, unsigned char, SPIClass*)'; libraries/Adafruit_BusIO/Adafruit_SPIDevice 2.cpp.o:/Users/antti/Documents/Arduino/libraries/Adafruit_BusIO/Adafruit_SPIDevice 2.cpp:17: first defined here
/Users/antti/Library/Arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: libraries/Adafruit_BusIO/Adafruit_SPIDevice.cpp.o: in function `Adafruit_SPIDevice::Adafruit_SPIDevice(signed char, signed char, signed char, signed char, unsigned long, BitOrder, unsigned char)':
/Users/antti/Documents/Arduino/libraries/Adafruit_BusIO/Adafruit_SPIDevice.cpp:43: multiple definition of `Adafruit_SPIDevice::Adafruit_SPIDevice(signed char, signed char, signed char, signed char, unsigned long, BitOrder, unsigned char)'; libraries/Adafruit_BusIO/Adafruit_SPIDevice 2.cpp.o:/Users/antti/Documents/Arduino/libraries/Adafruit_BusIO/Adafruit_SPIDevice 2.cpp:43: first defined here
/Users/antti/Library/Arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: libraries/Adafruit_BusIO/Adafruit_SPIDevice.cpp.o: in function `Adafruit_SPIDevice::Adafruit_SPIDevice(signed char, signed char, signed char, signed char, unsigned long, BitOrder, unsigned char)':
/Users/antti/Documents/Arduino/libraries/Adafruit_BusIO/Adafruit_SPIDevice.cpp:43: multiple definition of `Adafruit_SPIDevice::Adafruit_SPIDevice(signed char, signed char, signed char, signed char, unsigned long, BitOrder, unsigned char)'; libraries/Adafruit_BusIO/Adafruit_SPIDevice 2.cpp.o:/Users/antti/Documents/Arduino/libraries/Adafruit_BusIO/Adafruit_SPIDevice 2.cpp:43: first defined here
/Users/antti/Library/Arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: libraries/Adafruit_BusIO/Adafruit_SPIDevice.cpp.o: in function `Adafruit_SPIDevice::~Adafruit_SPIDevice()':
/Users/antti/Documents/Arduino/libraries/Adafruit_BusIO/Adafruit_SPIDevice.cpp:78: multiple definition of `Adafruit_SPIDevice::~Adafruit_SPIDevice()'; libraries/Adafruit_BusIO/Adafruit_SPIDevice 2.cpp.o:/Users/antti/Documents/Arduino/libraries/Adafruit_BusIO/Adafruit_SPIDevice 2.cpp:78: first defined here
/Users/antti/Library/Arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: libraries/Adafruit_BusIO/Adafruit_SPIDevice.cpp.o: in function `Adafruit_SPIDevice::~Adafruit_SPIDevice()':
/Users/antti/Documents/Arduino/libraries/Adafruit_BusIO/Adafruit_SPIDevice.cpp:78: multiple definition of `Adafruit_SPIDevice::~Adafruit_SPIDevice()'; libraries/Adafruit_BusIO/Adafruit_SPIDevice 2.cpp.o:/Users/antti/Documents/Arduino/libraries/Adafruit_BusIO/Adafruit_SPIDevice 2.cpp:78: first defined here
/Users/antti/Library/Arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: libraries/Adafruit_BusIO/Adafruit_SPIDevice.cpp.o: in function `Adafruit_SPIDevice::begin()':
/Users/antti/Documents/Arduino/libraries/Adafruit_BusIO/Adafruit_SPIDevice.cpp:85: multiple definition of `Adafruit_SPIDevice::begin()'; libraries/Adafruit_BusIO/Adafruit_SPIDevice 2.cpp.o:/Users/antti/Documents/Arduino/libraries/Adafruit_BusIO/Adafruit_SPIDevice 2.cpp:85: first defined here
/Users/antti/Library/Arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: libraries/Adafruit_BusIO/Adafruit_SPIDevice.cpp.o: in function `Adafruit_SPIDevice::transfer(unsigned char*, unsigned int)':
/Users/antti/Documents/Arduino/libraries/Adafruit_BusIO/Adafruit_SPIDevice.cpp:123: multiple definition of `Adafruit_SPIDevice::transfer(unsigned char*, unsigned int)'; libraries/Adafruit_BusIO/Adafruit_SPIDevice 2.cpp.o:/Users/antti/Documents/Arduino/libraries/Adafruit_BusIO/Adafruit_SPIDevice 2.cpp:123: first defined here
/Users/antti/Library/Arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: libraries/Adafruit_BusIO/Adafruit_SPIDevice.cpp.o: in function `Adafruit_SPIDevice::transfer(unsigned char)':
/Users/antti/Documents/Arduino/libraries/Adafruit_BusIO/Adafruit_SPIDevice.cpp:260: multiple definition of `Adafruit_SPIDevice::transfer(unsigned char)'; libraries/Adafruit_BusIO/Adafruit_SPIDevice 2.cpp.o:/Users/antti/Documents/Arduino/libraries/Adafruit_BusIO/Adafruit_SPIDevice 2.cpp:260: first defined here
/Users/antti/Library/Arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: libraries/Adafruit_BusIO/Adafruit_SPIDevice.cpp.o: in function `Adafruit_SPIDevice::beginTransaction()':
/Users/antti/Documents/Arduino/libraries/Adafruit_BusIO/Adafruit_SPIDevice.cpp:271: multiple definition of `Adafruit_SPIDevice::beginTransaction()'; libraries/Adafruit_BusIO/Adafruit_SPIDevice 2.cpp.o:/Users/antti/Documents/Arduino/libraries/Adafruit_BusIO/Adafruit_SPIDevice 2.cpp:271: first defined here
/Users/antti/Library/Arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: libraries/Adafruit_BusIO/Adafruit_SPIDevice.cpp.o: in function `Adafruit_SPIDevice::endTransaction()':
/Users/antti/Documents/Arduino/libraries/Adafruit_BusIO/Adafruit_SPIDevice.cpp:280: multiple definition of `Adafruit_SPIDevice::endTransaction()'; libraries/Adafruit_BusIO/Adafruit_SPIDevice 2.cpp.o:/Users/antti/Documents/Arduino/libraries/Adafruit_BusIO/Adafruit_SPIDevice 2.cpp:280: first defined here
/Users/antti/Library/Arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: libraries/Adafruit_BusIO/Adafruit_SPIDevice.cpp.o: in function `Adafruit_SPIDevice::setChipSelect(int)':
/Users/antti/Documents/Arduino/libraries/Adafruit_BusIO/Adafruit_SPIDevice.cpp:290: multiple definition of `Adafruit_SPIDevice::setChipSelect(int)'; libraries/Adafruit_BusIO/Adafruit_SPIDevice 2.cpp.o:/Users/antti/Documents/Arduino/libraries/Adafruit_BusIO/Adafruit_SPIDevice 2.cpp:290: first defined here
/Users/antti/Library/Arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: libraries/Adafruit_BusIO/Adafruit_SPIDevice.cpp.o: in function `Adafruit_SPIDevice::beginTransactionWithAssertingCS()':
/Users/antti/Documents/Arduino/libraries/Adafruit_BusIO/Adafruit_SPIDevice.cpp:301: multiple definition of `Adafruit_SPIDevice::beginTransactionWithAssertingCS()'; libraries/Adafruit_BusIO/Adafruit_SPIDevice 2.cpp.o:/Users/antti/Documents/Arduino/libraries/Adafruit_BusIO/Adafruit_SPIDevice 2.cpp:301: first defined here
[... cut ...] nt)':
/Users/antti/Documents/Arduino/libraries/Adafruit_BusIO/Adafruit_SPIDevice.cpp:485: multiple definition of `Adafruit_SPIDevice::write_and_read(unsigned char*, unsigned int)'; libraries/Adafruit_BusIO/Adafruit_SPIDevice 2.cpp.o:/Users/antti/Documents/Arduino/libraries/Adafruit_BusIO/Adafruit_SPIDevice 2.cpp:485: first defined here
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board Adafruit Feather nRF52840 Express.

### How to reproduce ?

Compile the schetch

hathach, 2022-10-18:
- please remove the error message, and upload the full compile log as attached txt file for readability
- Please specify the BSP (nRF52 core) version along with Adafruit_ILI9341, Adafruit_BusIO version. If they are not latest please update all involved libraries

I did an quick test, it seem to run just fine.

hathach, 2022-10-19:
that is ok, the compiled log include everything. The BSP is the core version, the one when you install/update via Board Manager. your is 1.3.0 which is already latest

```
adafruit/hardware/nrf52/1.3.0/cores/n
```

anttike, 2022-10-19:
It worked! Million thanks!

closed 2022-10-19 by anttike