adafruit/Adafruit_SPIFlash #139: Writing to FLASH-memory intermittently slow: 150us vs 80ms

tipo1000, 2022-11-25:
### Operating System

Windows 10

### IDE version

Arduino 1.8.13

### Board

Custom board based on SAMD21 (Adafruit Itsybitsy M0 express)

### BSP version

"Adafruit SAMD Boards" by Adafruit version 1.7.10

### SPIFlash Library version

3.11.0

### Sketch as attached file if not stock example

```
/*  Custom HW:
 *  Winbond 64Mbit FLASH- memory (W25Q64JVZPIQ) connected to ATSAMD21G via MUX-chip
 *  MUX is not switched during this test.
 *  Board selected in Arduino IDE: "Adafruit ItsyBitsy M0 Express (SAMD21)" (This board was used as a baseline for custom HW)
 */

#include <SPI.h>
#include <SdFat.h>
#include <Adafruit_SPIFlash.h>

#define FILE_NAME "Accdat_1.csv"            // Data is stored to this file

Adafruit_FlashTransport_SPI flashTransport(EXTERNAL_FLASH_USE_CS, EXTERNAL_FLASH_USE_SPI);
Adafruit_SPIFlash flash(&flashTransport);
FatFileSystem fatfs;

const int MUX_PIN = 7;                      // FLASH MUX pins
long startWriteTime;
long stopWriteTime;
long loopTime;
const int sampleCount = 2000;               // Number of "lines" written to FLASH -memory in this test.

void setup() {  
  Serial.begin(2000000); 
  delay(3000);
  while (!Serial) {
    delay(1);
  }
  Serial.println("Adafruit SPIFlash - memory writing test 1");
  Serial.println();

  SPI.begin();
  SPI.setDataMode(SPI_MODE0);  //lisätty

  // Initialize FLASH MUX pins.
  // Custom HW has MUX-chip in between ATSAMD21 and FLASH-memory chip
  // MUX is not switched during the test.
  pinMode(MUX_PIN, OUTPUT);
  digitalWrite(MUX_PIN, LOW);           //LOW = select FLASH0, HIGH = select FLASH1
  delay(1);
}

void loop() {
    // Initialize flash library and check its chip ID.
    if (!flash.begin()) {
      Serial.println("Error, failed to initialize flash chip!");
      while(1) yield();
[... 501 lines cut ...]
Sample #, loop time (us), write time (us): 379 , 1715 , 146
Sample #, loop time (us), write time (us): 380 , 1704 , 150
Sample #, loop time (us), write time (us): 381 , 1721 , 146
Sample #, loop time (us), write time (us): 382 , 1804 , 143
Sample #, loop time (us), write time (us): 383 , 86822 , 85204
Sample #, loop time (us), write time (us): 384 , 1739 , 145
Sample #, loop time (us), write time (us): 385 , 1815 , 143
Sample #, loop time (us), write time (us): 386 , 1872 , 148
Sample #, loop time (us), write time (us): 387 , 1988 , 146
Sample #, loop time (us), write time (us): 388 , 2009 , 146
Sample #, loop time (us), write time (us): 389 , 1860 , 149
Sample #, loop time (us), write time (us): 390 , 1890 , 149
```

Output showing data written to Flash:

```
Sample #, loop time (us), write time (us): 0 , 3206218 , 9652
0T3196568X1021120Y1000112Z100201
Sample #, loop time (us), write time (us): 1 , 3400 , 144
1T3209478X201121Y221221Z200211
Sample #, loop time (us), write time (us): 2 , 3016 , 144
2T3212496X1000212Y100111Z100111

```

hathach, 2022-11-25:
custom hardware is not supported since we don't have your hw to reproduce/test/fix the issue. 

PS: next time opening an issue, please attach sketch and log as txt for readibility.

closed 2022-11-25 by hathach

tipo1000, 2022-11-25:
Hi,

What do you mean by this: "attach sketch and log as txt"?
I copy-pasted the sketch and then marked it as "code" with "<>"-button.

Thanks,
Tipo

tipo1000, 2022-11-25:
I think I figured it out. I opened a new bug report where files are attached as txt.
The new bug report (#140) is based on standard Adafruit Itsybitsy M0 express (SAMD21) board.

 /tipo