adafruit/Adafruit_SPIFlash #144: Adafruit_SPIFlash library not working with Circuit Playground Express

MrPorcine, 2023-03-08:
### Operating System

Linux

### IDE version

Arduino 1.8.19

### Board

Circuit Playground Express

### BSP version

Adafruit Circuit Playground Express

### SPIFlash Library version

4.0.0

### Sketch as attached file if not stock example

Stock example "flash_info"

### What happened ?

load "flash_info" example. Compile fails with message "#error No (Q)SPI flash are defined for your board !". 
I see that there was a fix for this some time ago " Fixed for CPX #27 " Oct 2, 2019. Am I missing some #define?

### How to reproduce ?

Above.

hathach, 2023-03-09:
Just tested, and this indeed occurs when using Arduino one, but compiled fine with Adafruit core. Would you mind switching to use Adafruit core for now, we will try to either
- submit an PR to update define for cplay on Arduino core (will take time for them to merge and release)
- try to fix within the flash example.

hathach, 2023-03-10:
to use adafruit core, please follow this guide https://learn.adafruit.com/adafruit-arduino-ide-setup/arduino-1-dot-6-x-ide . Basically your need to
- add adafruit json link to IDE preference
- install adafruit samd package
- from IDE menu select "Adafruit SAMD" -> CircuitPlayground Express

Using adafruit package will also enable the use of TinyUSB and other libs from Adafruit ecosystem as well. Let me know if these work for you

[image]

MrPorcine, 2023-03-19:
Sorry for the late reply. I found the Arduino library SPIMemory that works with the Circuit Playground and fits my application very well. All I need is readByte, writeByte and eraseSector, so SPIMemory is a simple solution. Once you call the constructor with the select pin used on the CP, 

> SPIFlash flash(33) ;

 all works.

hathach, 2023-03-24:
no problems, I am glad you find your solution. We will leave this open until we fixed this though. Have fun.

PR #165 (2023-12-18): fix compile issue when building Circuit Playground express with Arduino core (not Adafruit core)
fix #144 compile issue when building Circuit Playground express with Arduino core (not Adafruit core). 
- this is due to The cpx variant in Arduino core lack the macro define for EXTERNAL_FLASH_USE_CS/EXTERNAL_FLASH_USE_SPI.
- Update flash_config.h to correct this case
closed 2023-12-18 by hathach