adafruit/Adafruit_SPIFlash #145: Cannot build Adafruit_SPIFlash library for ESP32

rspurlock, 2023-03-11:
### Operating System

Windows 11

### IDE version

1.8.19

### Board

Adafruit ESP32 Feather

### BSP version

1.0.6

### SPIFlash Library version

4.0.0

### Sketch as attached file if not stock example

examples/Adafruit ImageReader Library/EinkGray29BmpButtonDemo.ino

### What happened ?

Selecting a Adafruit ESP32 Feather Board and appropriate port try to build the example, fails with the following:

C:\Users\randy\Documents\Arduino\libraries\Adafruit_SPIFlash\src\esp32\Adafruit_FlashTransport_ESP32.cpp: In member function 'SPIFlash_Device_t* Adafruit_FlashTransport_ESP32::getFlashDevice()':
C:\Users\randy\Documents\Arduino\libraries\Adafruit_SPIFlash\src\esp32\Adafruit_FlashTransport_ESP32.cpp:62:3: error: 'esp_flash_t' was not declared in this scope
   esp_flash_t const *flash = _partition->flash_chip;
   ^
C:\Users\randy\Documents\Arduino\libraries\Adafruit_SPIFlash\src\esp32\Adafruit_FlashTransport_ESP32.cpp:63:36: error: 'flash' was not declared in this scope
   _flash_device.manufacturer_id = (flash->chip_id >> 16);
                                    ^
exit status 1
Error compiling for board Adafruit ESP32 Feather.

### How to reproduce ?

1. Simply attempt to build the example sketch
2. Commenting out the following lines in Adafruit_FlashTransport_ESP32.cpp allows the code to build:

In method Adafruit_FlashTransport_ESP32::getFlashDevice

  esp_flash_t const *flash = _partition->flash_chip;
  _flash_device.manufacturer_id = (flash->chip_id >> 16);
  _flash_device.memory_type = (flash->chip_id >> 8) & 0xff;
  _flash_device.capacity = flash->chip_id & 0xff;

In my case the code would then build and work properly. I attempted to find a definition for esp_flash_t in the ESP32 SDK in the installed 1.0.6 core but I could not find one. I checked the package used in my PlatformIO installation and it is version 2.0.6 and the ESP partition there *does* have a flash_chip (and corresponding esp_flast_t type) in it. I was unable to install that version in the Arduino IDE, it would only install 1.0.x versions.

hathach, 2023-03-13:
it compiles just fine in my PC here, arduino-esp32 1.0.6 is too old, you should try to install one of the latest arduino-esp32. Here is my Arduino IDE compile log which contains library information is used as follows

```
Using library Adafruit-GFX-Library at version 1.11.3 in folder: /home/hathach/Arduino/libraries/Adafruit-GFX-Library 
Using library Adafruit_BusIO at version 1.14.1 in folder: /home/hathach/Arduino/libraries/Adafruit_BusIO 
Using library Wire at version 2.0.0 in folder: /home/hathach/Arduino/hardware/espressif/esp32/libraries/Wire 
Using library SPI at version 2.0.0 in folder: /home/hathach/Arduino/hardware/espressif/esp32/libraries/SPI 
Using library Adafruit_EPD at version 4.5.0 in folder: /home/hathach/Arduino/libraries/Adafruit_EPD 
Using library SdFat_-_Adafruit_Fork at version 2.2.1 in folder: /home/hathach/Arduino/libraries/SdFat_-_Adafruit_Fork 
Using library Adafruit_ImageReader at version 2.8.1 in folder: /home/hathach/Arduino/libraries/Adafruit_ImageReader 
Using library Adafruit_SPIFlash at version 4.0.0 in folder: /home/hathach/Arduino/libraries/Adafruit_SPIFlash 
/home/hathach/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-size -A /tmp/arduino_build_989391/EInkGray29BmpButtonDemo.ino.elf
Sketch uses 307853 bytes (23%) of program storage space. Maximum is 1310720 bytes.
Global variables use 23624 bytes (7%) of dynamic memory, leaving 304056 bytes for local variables. Maximum is 327680 bytes.
```

closed 2023-03-13 by hathach

rspurlock, 2023-03-13:
For anyone else that might have this problem I had been using ESP chips for quite a while and didn't realize that they had changed the location of the ESP BSP packages and just had to update that in the boards manager preference for it to offer me the upgrade to the 2.0.7 BSP.

Thanks Ha!