adafruit/Adafruit_SPIFlash #155: Can not format 32MB chip (W25Q256JV)

ysnctr, 2023-09-26:
### Operating System

MacOS

### IDE version

32768

### Board

Feather NRF52840 Express (Raytac MDBT50Q with Adafruit nrf52840 express bootloader)

### BSP version

Adafruit Feather NRF52840 Express

### SPIFlash Library version

4.2.0

### Sketch as attached file if not stock example

examples/SdFat_format

### What happened ?

I run the same code with different SPI memories without any issues. However, the program does not format the chip W25Q256JV. It stucks inside the function "FRESULT r = f_mkfs("", FM_FAT, 0, workbuf, sizeof(workbuf));"  (an infinite loop). 

It maybe because of formatting the chip with fat12, although the chip size is 32MB. Hence, I changed the parameter of this function from FM_FAT to FS_FAT32, but still I have the same issue (infinite loop). (FRESULT r = f_mkfs("", FS_FAT32, 0, workbuf, sizeof(workbuf));)

Is there any solution to this issue? 

### How to reproduce ?

Run the SdFat_format code with W25Q256JV chip. 

### Debug Log as attached txt file

!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
This sketch will ERASE ALL DATA on the flash chip and format it with a new filesystem!
Type OK (all caps) and press enter to continue.
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
Creating and formatting FAT filesystem (this takes ~60 seconds)...  
(nothing else)

### Screenshots

[image]

TH1022, 2023-10-02:
same issue here

hathach, 2023-10-03:
where did you buy feather nRF52840 expres  ? the board only comes with 2MB flash

ysnctr, 2023-10-16:
Hi,

This is a custom board I designed, and it utilizes the Raytac MDBT50Q-1MV2 chip. For firmware updates, I've integrated the Adafruit NRF52840 Express Bootloader. In this design, I've incorporated the W25Q256JV flash memory. Previously, I tested the board with the W25Q128JV_PM, and the library worked as expected. However, I'm currently encountering an issue with the W25Q256JV.

It seems that the example code provided is written for FAT12 formatting, which is not compatible with the larger capacity of the W25Q256JV (32MB). Consequently, I'm unable to format it using the existing code.

If you have any insights or recommendations on how to format the chip as FAT32 or any other steps to overcome this challenge, I would greatly appreciate your assistance.

hathach, 2023-10-16:
custom board is not supported, we simply don't have the hardware to test/support. PR is welcome, next time, please fill in the correct board info in the issue template.

closed 2023-10-16 by hathach

ysnctr, 2023-10-16:
I understand that custom boards aren't officially supported, and I appreciate your response. While my board is indeed a custom design, it closely resembles the Adafruit NRF52840 Express. Given this similarity, I'm hoping for guidance on how to format the W25Q256JV chip as FAT32. 

If you can provide any instructions or insights on how to achieve this, it would be greatly appreciated. Thank you for your assistance.