adafruit/Adafruit_SPIFlash #158: Cannot run with custom SPI

hoanglam9925, 2023-10-14:
### Operating System

MacOS

### IDE version

1.18.9 and 2.2.1

### Board

esp32-s2-mini-1 (LOLIN S2 Mini)

### BSP version

esp32 v4.0.9

### SPIFlash Library version

4.3.0

### Sketch as attached file if not stock example

examples/flash_info

### What happened ?

i'm using esp32-s2 and w25q32/w25q128 connect to spi and CS pin is pin 2. I run examples/flash_info with CUSTOM_SPI and CUSTOM_CS, it auto reset at flash.begin()
if i comment CUSTOM_SPI and use default of library this return id is 0xfffff

### How to reproduce ?

[image]

PhongMaiAlterno, 2023-10-16:
i have same issue

hathach, 2023-10-16:
for esp32 port, this library is currently implemented to support only flash device that stores code since we don't make any hardwares that have 2nd flash dev. PR is welcome.

closed 2023-10-16 by hathach

leahcornelius, 2025-01-29:
So there is no reason external flash couldn't be used on esp32 with this code? Provided the defaulting to internal flash when ARDUINO_ARCH_ESP32 is defined is removed?