adafruit/Adafruit_SPIFlash #173: Winbond W25Q256JVFM flash parameters with ESP32 doubt

RcoreL, 2024-05-14:
Hello!

I am trying to connect the ESP32-WROOM-32UE with one winbond W25Q256JVFM flash using Dual SPI.
I need to fill these variables that appear on the adafruit library (flash_devices.h):

`#define W25Q256JV                                                              
  {                                                                            
    .total_size = (1UL << 25), /* 32 MiB */                                    
    .start_up_time_us = 5000, .manufacturer_id = 0xef,                     
    .memory_type = 0x70, .capacity = 0x19, .max_clock_speed_mhz = 133,         
    .quad_enable_bit_mask = 0x00, .has_sector_protection = true,              
    .supports_fast_read = true, .supports_qspi = true,                         
    .supports_qspi_writes = true, .write_status_register_split = false,        
    .single_status_byte = false, .is_fram = false,                             
  }`
I am pretty sure that connections are right because al receive some information of the flash (written below).

I look for all the data in the datasheet (https://www.winbond.com/resource-files/w25q256jv%20spi%20revg%2008032017.pdf) but it is not really clear. I need to know what to fill in all these gaps and prove if it is actually right with my flash model.

The only data i think is correct is: Manufacturer_id: 0xef / Memory_type: 0x70 /Capacity: 0x19. Because i use the program to ask it to the flash and it returns me this value. 

Can someone help me to find and how to know what to write in here? Have i to change anything else in these files?

Thank you so much for your help!

RcoreL, 2024-05-20:
I solved it. The problem was clock's speed, i was using 80 Mhz and i need to reduce it to 2-10 Mhz because of something i doesn't know yet.

Aditya-Shandilya, 2024-06-14:
Hey I am working on a similar project and trying to achieve a similar objective with my W25Q32. Could you please share your code as reference so that I can have a good starting point for my project

RcoreL, 2024-06-17:
Hi @Aditya-Shandilya , i am using the same code of the library with the same configuration as my first post and then i configure the clok speed to 5 MHz and it is working well. There's no more things in my code. 
Regards

hathach, 2024-06-17:
Thank you for the update, I don't have the chip,though 10Mhz is a bit too slow since esp32 spi can actually run 40/60mhz with ease. Maybe it is other issue, however, since we don't have the hardware/time to test this. I am glad you figured it out, any PR would be welcome.

closed 2024-06-17 by hathach

gabrielbravov, 2024-10-09:
Hello, i tried this configuration with ESP32 breakout board that has an esp-wroom-32 using W25Q128JVSIQ as external flash but didn't work with different frequency configurations. I'm still receiving JEDEC ID: 0xFFFFFF

How you did it work?