adafruit/Adafruit_SPIFlash #183: Library assumes onboard flash is always used on Pi Pico, prevents external memory from being used correctly.

Patzlyk, 2025-01-04:
### Operating System

Others

### IDE version

PlatformIO 3.3.3, VSCode 1.96.2

### Board

Raspberry Pi Pico W

### BSP version

Arduino-pico 4.4.1

### SPIFlash Library version

5.0.1

### Sketch as attached file if not stock example

[sketch.zip](https://github.com/user-attachments/files/18308589/sketch.zip)

### What happened ?

On line 64 of "Adafruit_SPIFlashBase.cpp", there is check for the RP2040 platform. If true, the onboard flash is automatically configured and the flash detection sequence is skipped. If an external SPI flash is being used, this assumption prevents it from being correctly recognized.
Disabling this check allows the library to work with external flash chips.
JEDEC ID, size, numPages and sectorCount are affected by this. Haven't tried writing or erasing yet.

Note this probably also happens on ESP32, however I could not test it.

### How to reproduce ?

1. Connect an external SPI flash module to the Pi Pico (W25QQ32BV)
2. Initialize the library
3. Try to read the JEDEC ID

The output should be wrong, unless the platform check on line 64 of "Adafruit_SPIFlashBase.cpp" is disabled.

### Screenshots

Wrong output:
[image]
Correct output:
[image]

e2forlife, 2025-01-07:
I ran into the same issue.  as a work around I commented out the Pico internal memory and let the lib code fall through to the non-pico variant.  Attached my "patched" version that appears to work, but I admit I haven't tested everything :)

in AdafruitSPIFlashBase.cpp add at line 64 `#if 0`
On line 91 add `#endif`

[Adafruit_SPIFlashBase.zip](https://github.com/user-attachments/files/18335051/Adafruit_SPIFlashBase.zip)

leahcornelius, 2025-01-29:
This issue is also present on ESP32 due to the ARDUINO_ARCH_ESP32 flag